jerryscript-project/jerryscript #3622: Question: Is there C API corresponds to `string.charCodeAt()`?

niklauslee, 2020-03-18:
Hello,

I'm writing an encoding function from string to Uint8Array based on `'ascii'` encoding (1 char to 1 byte).

I considered the functions `jerry_string_to_char_buffer()` and `jerry_string_to_utf8_char_buffer()`, but it encodes two or more bytes for exceeding ascii characters (0~127).

Here is an example of ascii encoding in Node.js:

```js
> var s = String.fromCharCode(0, 1, 127, 128, 255);
> var buf = new Buffer(s, 'ascii'); // <Buffer 00 01 7f 80 ff>
```

Is there any function or way to get char code at particular index from a string? (like `jerry_string_char_code_at()`).

Thank you in advance.

rerobika, 2020-03-26:
Hi @niklauslee!

The API does not contain directly any function to get a single character from a string, but you can use [jerry_substring_to_char_buffer](https://jerryscript.net/api-reference/#jerry_substring_to_char_buffer), with a `start = index` and `end = index + 1` arguments, to obtain a single character into a buffer. I hope it helps to resolve your problem.

niklauslee, 2020-03-27:
Actually I have resolved in a different way by making a function to convert utf-8 buffer to ascii buffer. Thank you @rerobika.

closed 2020-03-27 by niklauslee