jerryscript-project/jerryscript #3634: BUG: Number.toString([2|16])

cyongbin, 2020-03-21:
# JerryScript revision
The latest

# Case
```javascript
var i = 0;
for (i = 0; i < 20; i++) {
	console.log(`${i}.toString(16)=${i.toString(16)}`);
}

for (i = 0; i < 20; i++) {
	console.log(`${i}.toString(2)=${i.toString(2)}`);
}
```

# Output
```
0.toString(16)=0
1.toString(16)=1.0
2.toString(16)=2.0
3.toString(16)=3.0
4.toString(16)=4.0
5.toString(16)=5.0
6.toString(16)=6.0
7.toString(16)=7.0
8.toString(16)=8.0
9.toString(16)=9.0
10.toString(16)=a
11.toString(16)=b.0
12.toString(16)=c
13.toString(16)=d.0
14.toString(16)=e
15.toString(16)=f
16.toString(16)=10.0
17.toString(16)=11
18.toString(16)=12
19.toString(16)=13
0.toString(2)=0
1.toString(2)=1.0
2.toString(2)=10.0
3.toString(2)=11.0
4.toString(2)=100.0
5.toString(2)=101.0
6.toString(2)=110.0
7.toString(2)=111.0
8.toString(2)=1000.0
9.toString(2)=1001.0
10.toString(2)=1010
11.toString(2)=1011.0
12.toString(2)=1100
13.toString(2)=1101.0
14.toString(2)=1110
15.toString(2)=1111
16.toString(2)=10000.0
17.toString(2)=10001
18.toString(2)=10010
19.toString(2)=10011
```

# Bug
Some result of  `Number.toString([2|16])` end of `.0`.

akosthekiss, 2020-03-21:
@cyongbin Thanks for the bug report. But. Please do provide the requested details. "JerryScript revision - The latest" is not a useful, relevant, or future-proof information in the report.

akosthekiss, 2020-03-23:
"master head" is still only as good as "the latest". Tomorrow, it will refer to a different revision than today. Please, give the exact git hash you are using. (If you are working in a fork, then give the git hash of the base of the fork, i.e., mention a commit that is in the master branch of this repo.)

galpeter, 2020-03-23:
I've tried to reproduce the issue, however I was unable to get the same result (I've replaced the `console.log` calls with `print`. For me the output does not contain the extra `.0` characters . I've tried with multiple hash:
* 5b6d4f39b1b43907d8d60ea3b17bbf147f02415e  (the current master head at the time of this message.
* v2.2.0
* v2.1.0
* v2.0

Is it possible that you have modifications somewhere?

cyongbin, 2020-03-24:
This is very strange, we will continue to test, if there are any clues, I will give you feedback. 
Thank you very much

galpeter, 2020-03-24:
A few things which could help us track down the issue:
* What is the build configuration? (which parts are enabled/disabled?)
* What is the target platform (arm, x86, x64)?
* What is the compiler and it's version used?
* Are there any extra compiler options used?

cyongbin, 2020-03-26:
@galpeter We found that there was no problem with compiling on the arm platform, which was found in i386 gcc.

galpeter, 2020-03-26:
So the problem occurs with i386 gcc ? Could you share the compiler's version number?

rerobika, 2020-04-20:
@cyongbin Pinging the issue! Is there any progress about it?

cyongbin, 2020-04-27:
@rerobika It's a x86 GCC -O2 Error. We fixed this error by rewriting the function 'ecma_number_to_decimal ''.

```
/*
 * Fixed x86 GCC -O2 Error.
 */
#if defined(SYLIXOS) && defined(LW_CFG_CPU_ARCH_X86)
#if defined(__OPTIMIZE__) && __OPTIMIZE__ > 0

#include <math.h>

/**
 * 64 Bits integer to string.
 */
static int
lltoa (int64_t value, char *string, int radix)
{
  char tmp[32];
  char *tp = tmp;
  int i;
  uint64_t v;
  int sign;
  char *sp;

  if (radix > 36 || radix <= 1)
  {
    return 0;
  }

  sign = (radix == 10 && value < 0);
  if (sign)
    v = -value;
  else
    v = (uint64_t)value;
  while (v || tp == tmp)
  {
    i = v % radix;
    v = v / radix;
    if (i < 10)
      *tp++ = (char)(i + '0');
    else
      *tp++ = (char)(i + 'a' - 10);
  }

  sp = string;

  if (sign)
    *sp++ = '-';
  while (tp > tmp)
    *sp++ = *--tp;
  *sp = 0;

  return sp - string;
}

#endif /* __OPTIMIZE__ > 0 */
#endif /* SYLIXOS && LW_CFG_CPU_ARCH_X86 */

lit_utf8_size_t
ecma_number_to_decimal (ecma_number_t num, /**< ecma-number */
                        lit_utf8_byte_t *out_digits_p, /**< [out] buffer to fill with digits */
                        int32_t *out_decimal_exp_p) /**< [out] decimal exponent */
{
  JERRY_ASSERT (!ecma_number_is_nan (num));
  JERRY_ASSERT (!ecma_number_is_zero (num));
  JERRY_ASSERT (!ecma_number_is_infinity (num));
  JERRY_ASSERT (!ecma_number_is_negative (num));

#if defined(SYLIXOS) && defined(LW_CFG_CPU_ARCH_X86)
#if defined(__OPTIMIZE__) && __OPTIMIZE__ > 0

  double intpart;
  double fractpart;

  fractpart = modf ((double)num, &intpart);
  if (fractpart == 0.0)
  {
    int len = lltoa((int64_t)intpart, (char *)out_digits_p, 10);

    *out_decimal_exp_p = len;
    return len;
  }

#endif /* __OPTIMIZE__ > 0 */
#endif /* SYLIXOS && LW_CFG_CPU_ARCH_X86 */

  return ecma_errol0_dtoa ((double) num, out_digits_p, out_decimal_exp_p);
} /* ecma_number_to_decimal */
```

rerobika, 2020-04-27:
@cyongbin  Please provide **detailed** information about the running environment and the compiler version.

rerobika, 2020-05-26:
Closing the issue due to inactivity, please reopen if needed.

closed 2020-05-26 by rerobika