jerryscript-project/jerryscript #3646: An issue about Array.prototype.lastIndexOf

YuanWangC, 2020-03-26:
According to the description of `Array.prototype.lastIndexOf` in ES5.1 and ES6, when `fromIndex` is greater than or equal to the length of array, the whole array will be searched. But the output shows that the element can not be found.

###### JerryScript revision
[996bf76](https://github.com/jerryscript-project/jerryscript/commit/996bf76f5956c65342c1726d3f629fbea41320ab)

###### Test case
```
var NISLFuzzingFunc = function(){
    return [1].lastIndexOf(1, 1);
};
var NISLCallingResult = NISLFuzzingFunc();
print(NISLCallingResult);
```
###### Output
`-1`
###### Expected Output
`0`

galpeter, 2020-03-26:
The revision you mentioned is a bit old. On the current master this works as expected. Did a quick bisect and it seems that the 89db9253c8255dbb60e242b81ff40b0963af1361 fixed this.

rerobika, 2020-03-26:
The issue seems resolved to me, closing it.

closed 2020-03-26 by rerobika