jerryscript-project/jerryscript #3661: A question about this

YiWen-y, 2020-04-01:
When printing `this`, most engines output `[object global]`, such as v8, spiderMonkey, chakra, etc., but the output of jerry is `[object Object]`. Why does it happen?

###### JerryScript revision
[996bf76](https://github.com/jerryscript-project/jerryscript/commit/996bf76f5956c65342c1726d3f629fbea41320ab)

###### Build platform
    Ubuntu 16.04.6 LTS (Linux 4.15.0-55-generic x86_64)

###### Build steps
    python tools/build.py --clean --debug --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --mem-heap=1572864 --profile=es2015-subset

###### Test case
```
var f = function() {
    print(this);
};
f();
```

###### Execution steps
    build/bin/jerry testcase.js

###### Output
```
 [object Object]
```
###### Expected behavior
```
 [object global]
```

rerobika, 2020-04-01:
Hi @YiWen-y,

The `console.log`/`print` like functions are engine specific and the standard does not contain any point about them, so I would not say that this is the expected behavior only other engines creates different output.

rerobika, 2020-04-03:
The issue seems resolved, closing it.

closed 2020-04-03 by rerobika