jerryscript-project/jerryscript #3664: An issue about function.prototype.call

YiWen-y, 2020-04-02:
According to Standard ECMA-262(from ES3 to ES10), when running this testcase blow, undefined should be converted to the global object in line 3 and line5, but jerry didn't do that in line 5 and threw a TypeError: Argument cannot be converted to an object.

###### JerryScript revision
[996bf76](https://github.com/jerryscript-project/jerryscript/commit/996bf76f5956c65342c1726d3f629fbea41320ab)

###### Build platform
    Ubuntu 16.04.6 LTS (Linux 4.15.0-55-generic x86_64)

###### Build steps
    python tools/build.py --clean --debug --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --mem-heap=1572864 --profile=es2015-subset

###### Test case
```
var f = function() {
    print(typeof Object.prototype.isPrototypeOf);
    var a = Object.prototype.isPrototypeOf.call(undefined);
    print("a:", a);
    var b = Object.prototype.isPrototypeOf.call(undefined, []);
    print("b:", b);
};
f();
```

###### Execution steps
    build/bin/jerry testcase.js

###### Output
```
function
a: false
Exception backtrace:
    0: test case.js:5
    1: test case.js:8
Script Error: TypeError: Argument cannot be converted to an object.
```
###### Expected behavior
```
function
a: false
b: false
```

galpeter, 2020-04-03:
Thanks for the report. A couple of questions:
* What is your reference engine? I've tested this with node/ff/gjs and all of them reported the same TypeError.
* I've checked the standards (4->10) and if I read it correctly the `this` value should be converted via `ToObject` in case of the `isPrototypeOf` call . The `ToObject` should throw and error for `undefined` values. Could you point me to the standard where the `undefined` value should be converted to the global object?

YiWen-y, 2020-04-03:
Here is my understanding :
 `Object.prototype.isPrototypeOf` is a function, so when executing` Object.prototype.isPrototypeOf.call`, it should execute `Function.prototype.call` first, the ES standard  states that if the first parameter `thisArg` is undefined or null, it should be converted to a global object in note1 of [Function.prototype.call](http://www.ecma-international.org/ecma-262/10.0/index.html#sec-function.prototype.call).I wonder if my understanding is correct?

galpeter, 2020-04-03:
True, the `.call` is invoked, however based on the note 1 the `thisArg` is passed without modification (as per the first sentence). The second part mentions that in ES 3 the `thisArg` was converted if it was a `null` or `undefined` values. Based on this, the conversion is only in ES 3 and in ES 4 (and newer) this is not present. 

I've also checked this in the ES 3 standard:
"""
15.3.4.4 Function.prototype.call .... :
..
If thisArg is null or undefined,  the  called  function  is  passed  the  global  object  as  the this value.Otherwise, the called function is passed ToObject(thisArg) as the this value.
"""
src: https://www.ecma-international.org/publications/files/ECMA-ST-ARCH/ECMA-262,%203rd%20edition,%20December%201999.pdf  section: 15.3.4.4

This part is not present in newer standards.

YiWen-y, 2020-04-03:
If so, why is there no error when executing line 2 of `Object.prototype.isPrototypeOf.call (undefined)`?

galpeter, 2020-04-03:
That is because of the `isPrototypeOf`'s first step ( https://www.ecma-international.org/ecma-262/5.1/#sec-15.2.4.6 ):
"1. If V is not an object, return false."
Where `V` is the first argument of the `isPrototypeOf`.
In case of `Object.prototype.isPrototypeOf.call(undefined);` the `V` is not specified and it'll be `undefined`. Thus the step 1. will return false in the method.

In the second case: `Object.prototype.isPrototypeOf.call(undefined, []);` the `V` is the `[]` array which is an object and the step 1 will not return false.

YiWen-y, 2020-04-03:
Thank you so much for your patience, I misunderstood the standard before.

galpeter, 2020-04-03:
You are welcome.

Closing issue.

closed 2020-04-03 by galpeter