jerryscript-project/jerryscript #3674: DOS attack

rain6851, 2020-04-13:
#### JerryScript revision
76bbb088e91d36ee19349b06127eb0cbd5eea94f

###### Build platform
```
operating system: ubuntu18.04
```
###### Build steps
```
python tools/build.py --debug --logging=on --error-messages=on --line-info=on --jerry-debugger=on
```

###### Test case
```
try {
    alphabetR = ['a'];
    var ZHeb = alphabetR.push(alphabetR);
    var WsfW = alphabetR.toLocaleString();
} catch (ex) {
    sputnikException = ex;
}
var successfullyParsed = true;
```
###### Execution platform
operating system: ubuntu18.04

###### Execution steps
```
./build/bin/jerry poc
```

###### Backtrace
[image]

###### Expected behavior
Aborted (core dumped)

###### vulnerability description
After analysis, it is found that this is an assert. The key place for assert is in
[image]

Through the index, the conversion of the string is through the index, when encountering the object, the object will be found through the pointer. Because `alphabetR` pushed itself, creating a circular list, and then in `toLocaleString`,

[image]
Constantly looping to find its own object, causing the object to find a reference error, that is, `ERR_REF_COUNT_LIMIT`.

rerobika, 2020-04-14:
@rain6851 First of all, thanks for the report.

`ERR_REF_COUNT_LIMIT` is only a message about the reason of the incoming `SIGABRT`. Therefore the actual and the expected behaviors are the same. The reason of this abort, the infinite recursion in `Array.prototype.toLocalString`, which can be limited/prevented by using the `--stack-limit` build option.

rerobika, 2020-04-20:
@rain6851 Can we treat this issue resolved?

rain6851, 2020-04-20:
ok

rain6851, 2020-04-20:
> @rain6851 Can we treat this issue resolved?

ok

closed 2020-04-20 by rerobika