jerryscript-project/jerryscript #3691: ToLength function uses different limits than stated in ES6 standard

rwalczyna, 2020-04-22:
I have a question about ToLength function. I analyzed usage of Array.prototype.Map and it resulted in wrong output for lengths bigger than **2<sup>32</sup>**

###### Test code:
```js
function callbackfn(val, idx, obj) {
    print("Inside callback, idx:", idx);
    return val;
}
var obj = { 0: 5, 1: 13, length: "Infinity" };
var ret = Array.prototype.map.call(obj, callbackfn);
print(ret)
```
###### Output:
```
Inside callback, idx: 0
Inside callback, idx: 1
[application hangs]
```
###### Expected output:
```
RangeError
```

Application hangs, because JerryScript parses **Infinity** to **2<sup>32</sup>-1** and in internal functions it is just passed as *unit_32t* variable.

In [ES6 ToLength](http://www.ecma-international.org/ecma-262/6.0/#sec-tolength) function there is limit to **2<sup>53</sup> - 1** and it differs with [ES5.1 ToUint32](http://www.ecma-international.org/ecma-262/5.1/#sec-9.6) (which plays role of ToLength function).

I see that in the JerryScript both implementations use limits equal to **UINT32_MAX**:
[ecma-conversion.c:943](https://github.com/jerryscript-project/jerryscript/blob/1b1460e61fa468829984134c92761088d3cc8cbd/jerry-core/ecma/operations/ecma-conversion.c#L943).

I noticed that the same difference occurs in [String.prototype.split](https://github.com/jerryscript-project/jerryscript/blob/1b1460e61fa468829984134c92761088d3cc8cbd/jerry-core/ecma/builtin-objects/ecma-builtin-string-prototype.c#L795) and [RegExp.prototype [ @@split ]](https://github.com/jerryscript-project/jerryscript/blob/1b1460e61fa468829984134c92761088d3cc8cbd/jerry-core/ecma/operations/ecma-regexp-object.c#L1826)

Is there any specific reason why it is limited to 32 bits? Of course changing it would result in many modifications - because we would need to pass 64 bits variable in *ecma_op_to_length* function and change all uses.

Maybe it is made on purpose -  JerryScript aims to low memory devices - probably nobody would ever need that sizes of Arrays / Strings?

dbatyai, 2020-04-24:
We've had discussions about this issue a while ago, at the time we came to the conclusion that since we're targeting low-end devices, and 32-bit architectures in particular, it wouldn't make much sense to implement 64-bit lengths.

We could bring this up for debate again, to see if anything changed. I'm curious about what people think now.

IMO, for the reasons mentioned above, it still wouldn't be viable to implement, especially that I don't see a definite use case that would actually require lengths larger than uint32. Not to mention the fact that it would mess up a lot of structures internally, and as you mentioned, most built-in functions as well.

I'd rather stick with 32-bit lengths and be non-conforming than to force something that wouldn't be feasible.

ossy-szeged, 2020-06-11:
Just a note, related test262-es2015 tests:
built-ins/Array/prototype/map/15.4.4.19-3-14.js
built-ins/Array/prototype/map/15.4.4.19-3-28.js
built-ins/Array/prototype/map/15.4.4.19-3-29.js
built-ins/Array/prototype/map/15.4.4.19-3-8.js
built-ins/Array/prototype/slice/S15.4.4.10_A3_T1.js
built-ins/Array/prototype/slice/S15.4.4.10_A3_T2.js

rerobika, 2020-07-17:
Related PR: #4007

rerobika, 2020-07-29:
Resolved via #4007

closed 2020-07-29 by rerobika