jerryscript-project/jerryscript #3710: Assertion JERRY_CONTEXT (jmem_heap_allocated_size) == 0 in jmem_heap_finalize

renatahodovan, 2020-05-04:
###### JerryScript revision
6ff299c8

###### Build platform
Linux-4.15.0-88-generic-x86_64-with-Ubuntu-18.04-bionic

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```

###### Test case
```javascript
var obj = {
    get source() {
        return Set
    },
    [Symbol.match]: true
}

RegExp(obj)
```

###### Output
```text
ICE: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed at jerryscript/jerry-core/jmem/jmem-heap.c(jmem_heap_finalize):107.
Error: ERR_FAILED_INTERNAL_ASSERTION
ASAN:DEADLYSIGNAL
=================================================================
==37754==ERROR: AddressSanitizer: ABRT on unknown address 0x0000937a (pc 0xf7f95079 bp 0xff9d8bec sp 0xff9d8bd0 T0)
    #0 0xf7f95078  (linux-gate.so.1+0x1078)
    #1 0xf7f95078  (linux-gate.so.1+0x1078)
    #2 0xf77bc831 in raise (/lib/i386-linux-gnu/libc.so.6+0x2d831)
    #3 0xf77bdcc0 in abort (/lib/i386-linux-gnu/libc.so.6+0x2ecc0)
    #4 0x56638ec3 in jerry_port_fatal jerryscript/jerry-port/default/default-fatal.c:30
    #5 0x566ee88e in jerry_fatal jerryscript/jerry-core/jrt/jrt-fatals.c:63
    #6 0x566ee8cf in jerry_assert_fail jerryscript/jerry-core/jrt/jrt-fatals.c:87
    #7 0x566eda3e in jmem_heap_finalize jerryscript/jerry-core/jmem/jmem-heap.c:107
    #8 0x566ed864 in jmem_finalize jerryscript/jerry-core/jmem/jmem-allocator.c:170
    #9 0x56751ed7 in jerry_cleanup jerryscript/jerry-core/api/jerry.c:255
    #10 0x5674fb63 in main jerryscript/jerry-main/main-unix.c:994
    #11 0xf77a7e80 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18e80)
    #12 0x5662c0c0  (jerryscript/build/bin/jerry+0x1a0c0)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: ABRT (linux-gate.so.1+0x1078) 
==37754==ABORTING
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

rerobika, 2020-05-04:
Duplicate of #3671 , linked PR #3702 .

rerobika, 2020-05-14:
Closed via #3702

closed 2020-05-14 by rerobika