jerryscript-project/jerryscript #3737: Re-target JerryScript for ES.Next

akosthekiss, 2020-05-13:
Traditionally, JerryScript has been a fully ES5.1-compatible JavaScript engine. As the standard evolved, the community started adding features from the next edition of EcmaScript (ES6 or ES2015) into a new profile (configuration) of the engine. The original configuration was called the es5.1 profile, while the new one got the name es2015-subset profile. The original idea was to complete the ES2015 profile -- so as to be able to drop the "-subset" suffix -- and then move on to the next edition of the standard, and so on. I.e., the intent was to support all editions of the standard and allow our users to select the edition they need when they configure and build JerryScript.

It turns out that this is not a maintainable approach and does not even meet the needs of the users. Most JavaScript programmers use the latest edition of the standard, whichever that is, and expect engines to support that edition. Or, at least, support something close to that. Even if we finished the exact implementation of the ES6 specification, that would be both outdated and not even the subset of the latest edition of JavaScript. (Yes, there have been backward incompatible changes in the spec since 2015.) And we definitely wont be able to maintain different ES2015, ES2016, ES2017, ES2018, ES2019 (etc.) configurations. (Moreover, we couldn't even properly test conformance, as the test262 conformance suite has ceased releasing test suites for the different editions of the language after ES5.1.)

Therefore, I propose to keep the original ES5.1 profile (because that's a fully spec-compatible and mature configuration, often enough for constrained applications), but otherwise target the always-latest version of the EcmaScript standard (let's call it the ES.Next profile -- disclaimer: I haven't invented the term).

Notes:
- This would cause a big one-time churn in the code base, as all ES2015 macro guards would need a rename (to something like ESNEXT?). Most probably it's not even only the macros.
- This would cause some backward incompatibility as well, at least build system-wise, as profile name and feature guard names would change. Embedders relying on these would break.
- IMO, we still could and should keep the feature guards in the code base to allow users to trim down functionality. Let's remember that JerryScript is for constrained devices.

zherczeg, 2020-05-13:
I strongly support this idea. It is extremely hard to support multiple variants of the standard, especially at parser side.

bil-ash, 2020-05-14:
In case jerryscript moves on the path suggested by @akosthekiss please do make a final (github)release with ES2015-subset(including all the improvements since the last release) before moving on to ES.Next .

akosthekiss, 2020-06-12:
@bil-ash & all interested parties. The last version with the ES2015-subset profile just got released as JerryScript 2.3.0. Moreover, #3901 is already about to make to move to the new profile name and to the new standard target.

bil-ash, 2020-06-12:
Thanks, for considering my request. The 2.3.0 release will serve my purpose. Please do update kangax compatibility table(on the website) for 2.3.0 release,its seems to be still on 2.2.0 .

closed 2020-06-12 by dbatyai

ossy-szeged, 2020-06-22:
>  Please do update kangax compatibility table(on the website) for 2.3.0 release,its seems to be still on 2.2.0 (or may be put up the details somewhere on this repo of all the ES6 features supported in 2.3.0 release)

Of course we always update Kangax compatibilty table after the release as soon as possible. The update PR was already merged to the offical Kangax site 4 days after the release.

PR #3901 (2020-06-12): Re-target for ES.Next
A list of changes:
- 'es2015-subset' profile is deprecated, and an 'es.next' profile is added.
- The default profile is changed to 'es.next'
- Renamed the JERRY_ES2015 guard to JERRY_ESNEXT
- Renamed JERRY_ES2015_BUILTIN_* guards to JERRY_BUILTIN_*
- Moved es2015 specific tests to a new 'es.next' subdirectory
- Updated docs, targets, and test runners to reflect these changes

Resolves #3737.
commit fde0d556 (2020-06-12): Re-target for ES.Next (#3901)

A list of changes:
- 'es2015-subset' profile is deprecated, and an 'es.next' profile is added.
- The default profile is changed to 'es.next'
- Renamed the JERRY_ES2015 guard to JERRY_ESNEXT
- Renamed JERRY_ES2015_BUILTIN_* guards to JERRY_BUILTIN_*
- Moved es2015 specific tests to a new 'es.next' subdirectory
- Updated docs, targets, and test runners to reflect these changes

Resolves #3737.

JerryScript-DCO-1.0-Signed-off-by: Dániel Bátyai dbatyai@inf.u-szeged.hu