jerryscript-project/jerryscript #3753: stack-overflow in ecma_regexp_match

owl337, 2020-05-18:
###### JerryScript revision
bd1c4df

###### Build platform
Ubuntu 16.04.6 LTS (Linux 4.15.0-99-generic x86_64)

###### Build steps
python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --lto=off --error-message=on --system-allocator=on

###### Test case
r = new RegExp ("(()*?)+?a");
assert (r.exec("ba")[0] == "a");

###### Output
ASAN:SIGSEGV
=================================================================
==103435==ERROR: AddressSanitizer: stack-overflow on address 0xff318fcc (pc 0x0808fc96 bp 0xff319158 sp 0xff318fb0 T0)
    #0 0x808fc95 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:535
    #1 0x80915ee in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1060
    #2 0x80915ee in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1060
    #3 0x8091111 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:995
    #4 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #5 0x80915ee in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1060
    #6 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #7 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #8 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #9 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #10 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #11 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #12 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #13 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #14 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #15 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #16 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #17 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #18 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #19 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #20 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #21 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #22 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #23 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #24 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #25 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #26 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #27 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #28 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #29 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #30 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #31 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #32 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #33 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #34 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #35 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #36 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #37 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #38 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #39 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #40 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #41 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #42 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #43 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #44 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #45 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #46 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #47 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #48 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #49 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #50 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #51 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #52 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
[... 178 lines cut ...]
    #231 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #232 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #233 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #234 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #235 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #236 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #237 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #238 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #239 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #240 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #241 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #242 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #243 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #244 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #245 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #246 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #247 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #248 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #249 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122
    #250 0x8091b54 in ecma_regexp_match /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1122

SUMMARY: AddressSanitizer: stack-overflow /home/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:535 ecma_regexp_match
==103435==ABORTING

Credits: This vulnerability is detected by chong from OWL337.

zherczeg, 2020-05-18:
I think these reports need to be improved. Backtrace needs to be shorter, since it takes hours to scroll down to the bottom. If I want a backtrace I can produce it myself, so it can even be omitted from the report.

In my experiences 99% of these reports are belongs to the following categories:
- inifinite JS loops, recursions (fixing them is outside the scope of the engine)
- obsolote asserts (just improve or move the assert)
- trivial unhandled cases

Creating patches for these should take less than 3 minutes, so it would be good to replace the backtrace with an initial patch proposal. Thank you for your efforts.

akosthekiss, 2020-05-18:
@zherczeg You (we) cannot and must not expect those who are not active developers of the project to understand the code base and submit a PR to fix the issues they find. That's not their task.

zherczeg, 2020-05-18:
When real people submit issues, they try to be helpful, and often analize their own issue quite well, which shows their respect to the people who is working on the project. This feels missing from these automated reports. If the reporter would check the source code he sent, the problem would be obvious to him, and we could discuss it. There is no "fix" to infinite recursion in any language, only various ways to abort the execution.

dbatyai, 2020-05-18:
This pattern should not even result in infinite recursion, it is actually a problem with the current implementation. #3746 will resolve this however.

akosthekiss, 2020-05-18:
@zherczeg Please, don't do this. This is not even passive-aggressive, but you are actively accusing our community members of being disrespectful and not being helpful. None of which they are.

_Some_ members of the community have time, skills, and/or resources to develop the code base of the project. Some others may not be developers of the project though, but their feedback is still valuable. You simply MUST NOT alienate those who report issues in JerryScript. We have to be thankful for them for their time spent on helping us make JerryScript better by spotting faults _we_ missed.

And the above is also true for automated reports (even if I don't think that the above report was made by an automated system -- only found by one, perhaps). Someone has spent their time to create that automated system and ran it on JerryScript, for our benefit. (BTW, automation is not evil. It helps. CI is also our friend. They all help to focus human resources on the creative part of the development by lifting the burden of repetitive tasks from our shoulders.)

Having said all that, it may still happen that an issue report (be it "hand-made" or automatic) is invalid. Then, you have several options:
- You MAY ignore those issues that irritate you for other maintainers to handle.
- You may explain why you think an issue report is invalid. And you may also _suggest_ ways how to avoid them. E.g., how to build JerryScript with stack limit checks enabled.

The bottom line is, please, keep this project a welcoming and safe place for all current and potential members of our community.

@owl337 / chong: Thanks for your report. As @dbatyai mentioned, it is being fixed.

rerobika, 2020-05-26:
Resolved via #3746.

closed 2020-05-26 by rerobika