jerryscript-project/jerryscript #3763: JerryScript's execution result is inconsistent with other engines about 'new Uint8Array()'.

NWU-NISL, 2020-05-19:
When I ran the following testcase, jerry showed different results from the other engines.

###### Test case
```javascript
var a = new Uint8Array(undefined);
```

###### JerryScript revision
bd1c4df (committed 3 days ago)

###### Build Platform
Ubuntu 16.04
(Linux 3.10.0-957.12.1.el7.x86_64 x86_64)

###### Execution steps
>build/bin/jerry testcase.js

###### Jerry Output
>Script Error: TypeError: length argument is undefined

###### Other Engines Output
>No exception is thrown

I know jerryscript does not support all es6 standards, so I want to know if this is an unimplemented feature or a bug? Looking forward your reply, thanks!

galpeter, 2020-05-19:
Hi!

The es6 standard states that if the "length" argument is "undefined" a TypeError should be thrown ( http://www.ecma-international.org/ecma-262/6.0/#sec-%typedarray%-length 22.2.1.2 step 3). However, in newer standards (es 8) this was reworked and the "length" is calculated via the "ToIndex" operation which handles the "undefined" argument by returning 0 as the length ( http://www.ecma-international.org/ecma-262/8.0/#sec-typedarray-length 22.2.4.2 step 3).

So, in light of this JerryScript is ES 6 compatible in this case. The other engines however implements newer standards this gives the different output.

NWU-NISL, 2020-05-20:
@galpeter I get it, thanks for your reply. 😄 
Closed.

closed 2020-05-20 by NWU-NISL