jerryscript-project/jerryscript #3783: Stack Exhaustion (ecma_pointer_to_ecma_value)

nszetei, 2020-05-22:
###### JerryScript revision
https://github.com/jerryscript-project/jerryscript/commit/6cd309b5252d15f3e3e6481c72a4d8fa7f9557a9

###### Build platform
Ubuntu 20.04 LTS (Linux 5.4.0-31-generic x86_64)

###### Build steps

```sh
python tools/build.py --profile=es2015-subset --lto=off --compile-flag=-g --error-messages=on --debug --compile-flag=-g --strip=off --logging=on --compile-flag=-fsanitize=address
```

###### Test case

```
var v2 = Symbol.iterator;
var v3 = "description"[v2];
var v7 = {get:v3};
var v9 = Object.defineProperty(Object,"toString",v7);
var v11 = [-486477.70497889485,Object,-486477.70497889485,"undefined"];
var v13 = Function(v11);
```

###### Output

```
$ build/bin/jerry /tmp/test.js
AddressSanitizer:DEADLYSIGNAL
=================================================================
==213194==ERROR: AddressSanitizer: stack-overflow on address 0x7ffe69638ff8 (pc 0x56290e120f02 bp 0x7ffe69639020 sp 0x7ffe69639000 T0)
    #0 0x56290e120f01 in ecma_pointer_to_ecma_value /home/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:95
    #1 0x56290e120f01 in ecma_make_object_value /home/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:663
    #2 0x56290e15a198 in ecma_op_general_object_default_value /home/jerryscript/jerry-core/ecma/operations/ecma-objects-general.c:236
[ .. snip .. ]
    #267 0x56290e15a3e9 in ecma_op_general_object_ordinary_value /home/jerryscript/jerry-core/ecma/operations/ecma-objects-general.c:311
    #268 0x56290e15a31d in ecma_op_general_object_default_value /home/jerryscript/jerry-core/ecma/operations/ecma-objects-general.c:289
```

akosthekiss, 2020-05-23:
Could you please try whether you get SO even if you set a limit for the stack? (Note: You can use `--stack-limit` option when building the engine to limit the maximum amount of stack that the engine can use.)

nszetei, 2020-05-23:
It looks that it's handled by the engine as I got:

```
Script Error: RangeError: Maximum call stack size exceeded.
```

For the future in this specific scenario with SO bugs, should I report them only when they are triggerable with the `stack-limit`?

akosthekiss, 2020-05-23:
Yes, IMO, SO is only a bug if it can be hit with `--stack-limit` enabled.

nszetei, 2020-05-23:
OK, I appreciate you letting me know and I am closing this issue.

closed 2020-05-23 by nszetei