jerryscript-project/jerryscript #3785: Stack Exhaustion (ecma_proxy_object_get, ecma_proxy_object_set)

nszetei, 2020-05-23:
###### JerryScript revision
https://github.com/jerryscript-project/jerryscript/commit/6cd309b5252d15f3e3e6481c72a4d8fa7f9557a9

###### Build platform
Ubuntu 20.04 LTS (Linux 5.4.0-31-generic x86_64)

###### Build steps

```sh
python tools/build.py --profile=es2015-subset --lto=off --compile-flag=-g \
--error-messages=on --debug --compile-flag=-g --strip=off --logging=on \
 --compile-flag=-fsanitize=address
```

###### Test cases

```
var v2 = {};
var v4 = new Proxy(Uint8Array,v2);
v4.__proto__ = v4;
v4[1] = 2;
```

```
var v1 = {};
var v3 = new Proxy(parseFloat,v1);
v3.__proto__ = v3;
var v6 = "aa".constructor;
var v7 = parseFloat & v6;
```

akosthekiss, 2020-05-23:
Could you please try whether you get SO even if you set a limit for the stack? (Note: You can use `--stack-limit` option when building the engine to limit the maximum amount of stack that the engine can use.)

nszetei, 2020-05-23:
> Could you please try whether you get SO even if you set a limit for the stack? (Note: You can use `--stack-limit` option when building the engine to limit the maximum amount of stack that the engine can use.)

Yes. Unlike https://github.com/jerryscript-project/jerryscript/issues/3783 here I got SO in both cases (e.g. with `--stack-limit=10`).

nszetei, 2020-05-23:
Just to have it grouped together, a PoC for `ecma_proxy_object_has`:

```
function main() {
var v1 = [13.37,13.37,13.37,13.37];
var v4 = {isExtensible:Infinity};
var v6 = new Proxy(WeakMap,v4);
v6.__proto__ = v1;
v1.__proto__ = v6;
with (v1) {
    valueOf = 0;
}
}
main();
```

PR #3796 (2020-05-26): Add stack limit check to proxy operations
Fixes #3785.
commit 15629e8b (2020-05-26): Add stack limit check to proxy operations (#3796)

Fixes #3785.

JerryScript-DCO-1.0-Signed-off-by: Dániel Bátyai dbatyai@inf.u-szeged.hu
closed 2020-05-26 by zherczeg