jerryscript-project/jerryscript #3830: AddressSanitizer: while reporting a bug found another one.

owl337, 2020-06-02:
###### JerryScript revision
d06c3a7

###### Build platform
Ubuntu 16.04.6 LTS (Linux 4.15.0-99-generic x86_64)

###### Build steps
```
 ./tools/build.py --clean --debug --compile-flag=-fsanitize=address \ 
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
 --compile-flag=-fno-common --compile-flag=-g --strip=off \
 --system-allocator=on --logging=on \
 --linker-flag=-fuse-ld=gold --error-messages=on \
 --profile=es2015-subset --lto=off --stack-limit=50 
```

###### Test case
```
str = '';
 for (j = 0; j < 5; j++)
{
  str += 'A';

  b: for (i = 0; i < 5; i++)
  {
    str += 'B';

    switch (1)
    {
      case 1:
        continue b;
      function f2()
    {
      assert(k > 0);
      assert(i < 10000);

      if(--k == 0)
        {
          return;
        }

      f2();
    }

  k = 17;

  f2();
    }

    str += 'C';
  }

  str += 'D';
}
```

###### Output
```
ASAN:SIGSEGV
=================================================================
==35306==ERROR: AddressSanitizer: SEGV on unknown address 0x00001c25 (pc 0x00001c25 bp 0x00001c25 sp 0xfff1b9a0 T0)
ASAN:SIGSEGV
==35306==AddressSanitizer: while reporting a bug found another one. Ignoring.
```

Credits: This vulnerability is detected by chong from OWL337.

galpeter, 2020-06-02:
Minimal test case:
```js
for (var i = 0; i < 2; i++)
{   
    switch (1)
    { 
      case 1:
        function f2() { f2(); }
        continue;
    }
}

```

rerobika, 2020-06-02:
Hint after some investigation:

The VM stack layout before the strict equal comparison:

```
1
block context
1
```
Therefore the strict equal comparison takes the top 2 elements from the stack which is incorrect. So a statement level block context is created **not** in the top of the registers end.

zherczeg, 2020-06-04:
I cannot reproduce the bug with any of the test cases above. Also the vm specifically handles this case:

https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/vm/vm.c#L3309

Apart from the indentation of the comment, I don't see anything wrong with it. Can somebody explain me how to reproduce this bug exactly?

commit 10e78a4e (2020-06-05): Context break should honor private contexts. (#3852)

Fixes #3830

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com
closed 2020-06-05 by dbatyai