jerryscript-project/jerryscript #3833: run build.py failed on windows10

drifting1024, 2020-06-03:
First of all, when reporting a bug, give the issue a descriptive title.

In the body of the issue, optionally give a free-form text description of the
bug. Give the context necessary for others to understand the problem.

Then, provide information necessary to reproduce the bug.
Omit sections that are irrelevant for the bug report, but note that information
like git revision, build platform, build command, and test case are required in
almost all cases.

###### JerryScript revision
a56e31f194fe52000ed95605ae3c92390cd60ad5

###### Build platform
Windows-10-10.0.18362-SP0

###### Build steps
```sh
cd jerryscript
python tools/build.py
```

###### Build log

###### Test case

###### Execution platform

###### Execution steps

###### Output
Traceback (most recent call last):
  File "tools/build.py", line 295, in <module>
    main()
  File "tools/build.py", line 282, in main
    ret = configure_jerry(arguments)
  File "tools/build.py", line 254, in configure_jerry
    return subprocess.call(cmake_cmd)
  File "C:\Users\EDZ\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Users\EDZ\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\EDZ\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] ϵͳ▒Ҳ▒▒▒ָ▒▒▒▒▒ļ▒▒▒

rerobika, 2020-06-03:
cc @ossy-szeged.

ossy-szeged, 2020-06-03:
It's not a bug. It seems you don't have CMake installed on your system. (and maybe Visual Studio is missing too)

drifting1024, 2020-06-03:
ok. i'll check

drifting1024, 2020-06-05:
i installed CMake 3.17.3, it changs nothing, still output same failed msg

drifting1024, 2020-06-05:
sorry, my fault.
i used 'git bash' to run the cmd instead of 'powershell'.
it works well with 'powershell'.
don't know why, anyway, it may be useful when you guys come across the same situation the next time

closed 2020-06-05 by drifting1024