jerryscript-project/jerryscript #3886: Reaching ERR_REF_COUNT_LIMIT lead to a crash in ecma_ref_object

ArayzWang, 2020-06-10:
###### JerryScript revision
ba4e3a402fce722ea752243c60c009307438d97f

###### Build platform
Ubuntu 18.04 LTS

###### Build steps
python tools/build.py --profile=es2015-subset --lto=off --error-messages=on --strip=off --compile-flag=-fsanitize=address

###### Test case
```
function main() {
function v0() {
    const v1 = new v0();
}
const v2 = v0();
}
main();
```
###### Execution steps
build/bin/jerry testcase.js

###### Output
Program received signal SIGABRT, Aborted.

###### Backtrace
Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
(gdb) bt
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff6e43801 in __GI_abort () at abort.c:79
#2  0x00000000005acae2 in jerry_port_fatal ()
#3  0x000000000053bedf in jerry_fatal ()
#4  0x00000000004f90d3 in ecma_ref_object ()
#5  0x00000000005042cd in ecma_copy_value ()
#6  0x000000000055cfca in vm_loop ()
#7  0x000000000055b5e6 in vm_execute ()
#8  0x000000000055b183 in vm_run ()
#9  0x000000000051f650 in ecma_op_function_call_simple ()
#10 0x000000000051fafa in ecma_op_function_construct ()
#11 0x000000000055bd22 in vm_execute ()
#12 0x000000000055b183 in vm_run ()
#13 0x000000000051f650 in ecma_op_function_call_simple ()
#14 0x000000000051fafa in ecma_op_function_construct ()
#15 0x000000000055bd22 in vm_execute ()
#16 0x000000000055b183 in vm_run ()
#17 0x000000000051f650 in ecma_op_function_call_simple ()
#18 0x000000000051fafa in ecma_op_function_construct ()
#19 0x000000000055bd22 in vm_execute ()
#20 0x000000000055b183 in vm_run ()
#21 0x000000000051f650 in ecma_op_function_call_simple ()
#22 0x000000000051fafa in ecma_op_function_construct ()
#23 0x000000000055bd22 in vm_execute ()
#24 0x000000000055b183 in vm_run ()
……
```

akosthekiss, 2020-06-10:
@ArayzWang Thanks for the report. I'm not sure it can be flagged as a valid bug, though. `ERR_REF_COUNT_LIMIT` signals that you hav reached the internal limits of the engine. Most probably because of the infinite recursion on `v0`.

(I'm not completely sure why this didn't cause a stack overflow instead, but perhaps simply the reference count limit was hit sooner. BTW, it could be useful if you enabled stack limit in your build. Then, infinite recursions will / should be turned into runtime exceptions.)

ArayzWang, 2020-06-10:
@akosthekiss I tried rebuild project with following arguments:python tools/build.py --profile=es2015-subset --lto=off --error-messages=on --strip=off --compile-flag=-fsanitize=address --stack-limit=3

ArayzWang, 2020-06-10:
@akosthekiss I tried rebuild with following arguments:
python tools/build.py --profile=es2015-subset --lto=off --error-messages=on --strip=off --compile-flag=-fsanitize=address --stack-limit=10
(Adding a ```--stack-limit=10``` to enable stack limit)
The crash still reproduce, is there any other way to enable stack limit?

akosthekiss, 2020-06-10:
@ArayzWang Nah, that should be the proper way. Interesting. Someone might want to take a look at this then. @zherczeg @rerobika perhaps?

rerobika, 2020-06-10:
I'll take a look at it.

rerobika, 2020-06-12:
Since this recursive function is quite "small" and the engine is compiled in release mode the internal refcount error is still reached earlier than the 10 KB stack limit. Reducing further the stack limit to 1 KB, the expected RangeError is thrown.

rerobika, 2020-06-15:
IMO we can treat this issue resolved, please reopen it if needed.

closed 2020-06-15 by rerobika