jerryscript-project/jerryscript #3894: Incorrect await behavior?

zealotds, 2020-06-11:
###### JerryScript revision
commit: cae6cd08fb3dc8d2ded58177f997f8e3b3747b28

###### Build platform
Debian GNU/Linux 10 (buster) (Linux 4.19.0-8-amd64 x86_64)

###### Build steps
`python tools/build.py --profile=es2015-subset`

###### Test case
Run following code with the CLI bin/jerry:
```
async function foo (promise) { let v = await promise; print('value:', v); }
let r = null
let bar = new Promise(resolver => r = resolver)
foo(bar)
```

###### Output
`value: [object Promise]`

###### Expected behavior
It should not print until `r('some value')`

zherczeg, 2020-06-11:
Await has not been implemented yet, currently a nop. I am working on the implementation.

clarkttfu, 2020-06-11:
Looking forward to it so much! Hope to hear you soon 👍

zealotds, 2020-06-11:
Looking forward to it so much! Hope to hear you soon 👍

closed 2020-06-17 by zherczeg

clarkttfu, 2020-06-17:
Excellent !!