jerryscript-project/jerryscript #3903: An issue about String.prototype.indexOf and String.prototype.lastIndexOf

NWU-NISL, 2020-06-15:
According to ES6.0, the return value of`String.prototype.indexOf(searchString, fromIndex)` is the smallest possible integer which is not smaller than `fromIndex` . And When executing this testcase on other engines, (like chakraCore, v8 and spiderMonkey) I get the expected output.  A similar situation also occurs when using `String.prototype.lastIndexOf (searchString [, position] )`. When executing testcase2, the output of other engines are all `1`, but the output of jerry is `5`.

#### version

`996bf76`

#### testcase1

```javascript
var NISLFuzzingFunc = function(){
    var a = "abcde".indexOf("",1);
    print(a);
};
NISLFuzzingFunc();
```

#### output1

```
0
```

#### expected output1

```
1
```

#### testcase2

```javascript
var NISLFuzzingFunc = function () {
    var a = 'abcde'.lastIndexOf("", 1);
    print(a);
};
NISLFuzzingFunc();
```

#### output2

```
5
```

#### expected output2

```
1
```

Contributor:[@YuanWangC](https://github.com/YuanWangC); [@Wen Yi](https://github.com/YiWen-y)

galpeter, 2020-06-15:
The first test case seems to be working as expected on the current master (it was probably fixed in f13e8b602e20a80bc7e2e32a971307c0d34e07ca 1 month old).

If possible please report bugs against a fairly recent version of the engine.

The second test case is a valid issue.

akosthekiss, 2020-06-15:
996bf76 is from last November. That means the reported revision is almost 7 months old! Do update your local repo to the latest origin master regularly, as suggested by @galpeter .

ossy-szeged, 2020-09-29:
Already fixed by 1be9573925ff3944031bb12d28ec4abb8c82593d ( #3978)

closed 2020-09-29 by ossy-szeged