jerryscript-project/jerryscript #3931: Implemented Fuzzilli Integration

nszetei, 2020-06-25:
Hello. I am creating a new issue as I haven't found a better contact. I implemented an integration for JerryScript for Fuzzilli fuzzer https://github.com/googleprojectzero/fuzzilli/pull/79 . It contains also an easy-to-use docker container or it could be invoked directly, as described in the [readme](https://github.com/googleprojectzero/fuzzilli/tree/master/Targets/Jerryscript) file. If you decide to give it a try, I believe it could still find some valuable bugs.

dbatyai, 2020-06-25:
Sounds great!

We already have a separate main for libfuzzer, I belive we would have no objection to add a separate main for fuzzilli as well, if you're willing to open a pull request. Instead of relying on a patch file, this would make sure that it's always up to date, as long as it does not require extensive amounts of maintenance.

nszetei, 2020-06-26:
Hey, I created a pull request [here](https://github.com/jerryscript-project/jerryscript/pull/3941).

closed 2020-07-23 by nszetei