jerryscript-project/jerryscript #3977: Stack overflow in ecma_is_lexical_environment

Changochen, 2020-07-06:
### JerryScript revision
git hash: 392ee717128f6daaf0b94e54fc9512eb26d680e2

### Test case
```javascript
function a() { new new Proxy(a, {}) }
JSON.parse("[]", a)
```

### Execution steps
./jerry poc.js

### Build cmd
python tools/build.py --compile-flag="-fsanitize=address"

Stack dump:
---
```
ASAN:DEADLYSIGNAL
=================================================================
==180140==ERROR: AddressSanitizer: stack-overflow on address 0x7fff1ed99fe8 (pc 0x5632f4db5751 bp 0x7fff1ed9a000 sp 0x7fff1ed99fe0 T0)
    #0 0x5632f4db5750 in ecma_is_lexical_environment /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/base/ecma-helpers.c:174
    #1 0x5632f4db5c78 in ecma_get_lex_env_type /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/base/ecma-helpers.c:274
    #2 0x5632f4ca68fa in ecma_op_resolve_reference_value /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-reference.c:276
    #3 0x5632f4d219de in vm_loop.lto_priv.485 /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:1036
    #4 0x5632f4cf8723 in vm_execute /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4670
    #5 0x5632f4cf8d17 in vm_run /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4778
    #6 0x5632f4da077e in ecma_op_function_call_simple /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:942
    #7 0x5632f4da17ea in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1366
    #8 0x5632f4d7a201 in ecma_proxy_object_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-proxy-object.c:1779
    #9 0x5632f4da1459 in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1264
    #10 0x5632f4d20f37 in opfunc_construct.lto_priv.481 /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:863
    #11 0x5632f4cf87d8 in vm_execute /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4697
    #12 0x5632f4cf8d17 in vm_run /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4778
    #13 0x5632f4da077e in ecma_op_function_call_simple /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:942
    #14 0x5632f4da17ea in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1366
    #15 0x5632f4d7a201 in ecma_proxy_object_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-proxy-object.c:1779
    #16 0x5632f4da1459 in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1264
    #17 0x5632f4d20f37 in opfunc_construct.lto_priv.481 /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:863
    #18 0x5632f4cf87d8 in vm_execute /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4697
    #19 0x5632f4cf8d17 in vm_run /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4778
    #20 0x5632f4da077e in ecma_op_function_call_simple /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:942
    #21 0x5632f4da17ea in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1366
    #22 0x5632f4d7a201 in ecma_proxy_object_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-proxy-object.c:1779
    #23 0x5632f4da1459 in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1264
    #24 0x5632f4d20f37 in opfunc_construct.lto_priv.481 /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:863
    #25 0x5632f4cf87d8 in vm_execute /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4697
    #26 0x5632f4cf8d17 in vm_run /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4778
    #27 0x5632f4da077e in ecma_op_function_call_simple /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:942
    #28 0x5632f4da17ea in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1366
    #29 0x5632f4d7a201 in ecma_proxy_object_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-proxy-object.c:1779
    #30 0x5632f4da1459 in ecma_op_function_construct /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/ecma/operations/ecma-function-object.c:1264
    #31 0x5632f4d20f37 in opfunc_construct.lto_priv.481 /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:863
    #32 0x5632f4cf87d8 in vm_execute /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4697
    #33 0x5632f4cf8d17 in vm_run /home/yongheng/SQLFuzzer/jerry_clean/jerry-core/vm/vm.c:4778
```
---

rerobika, 2020-07-06:
Hi @Changochen!

That's a general stack-overflow issue. The engine has no internal stack-overflow protection by default, but you can enable it by passing `--stack-limit X` to `build.py` where `X` is the limit of stack usage in KB. Whenever this limit is reached a range error is thrown.

closed 2020-07-06 by rerobika

vinay54321, 2024-09-03:
Can you tell what value can we use 32bit linux based machine please.