jerryscript-project/jerryscript #3984: questions about the heap allocator

godmiaozi, 2020-07-07:
I have several questions about the heap allocator which jerry is using, hope someone can help.

* the allocator have 2 different strategies, one is for 8bytes chunk allocation, the other is for large size chunk allocation. why jerry choose 8bytes alignment， not 4, 5 or 7. Is there any special consideration about this?
*  In some scenario, i have found some memory fragment issue with jerry. In the future is there any plan that a compacting  gc algorithm will be implemented?

rerobika, 2020-07-07:
cc: @dbatyai

dbatyai, 2020-07-07:
Hi @godmiaozi,

5 or 7 byte alignment isn't very feasible, memory should always be aligned to a power of two to be efficient. The main deciding factor between 4 and 8 byte alignment was how jerryscript allocates memory internally. Jerryscript used to allocate a lot of 8 byte sized chunks (8 byte chunks are still the most common, but less so than in the past), so it made sense to choose 8 byte alignment. Using 4 byte alignment would've also meant that we would have to align memory manually after allocation for values that require it (like doubles), which would've wasted additional memory and potentially caused further fragmentation.

We've also seen that some workloads can cause excessive fragmentation. Unfortunately a compacting gc is not feasible at this point due to stray pointers on the C stack, which prevent us from safely moving objects. We would have to transition to a handle based approach, but that would have rather large impact on memory usage, so I'm not sure whether it would be worth the effort.

There are plans however to improve the allocation strategy to cause less fragmentation, and also improve the algorithm so that fragmentation has less of an effect on performance. I can't give an ETA at this point, but it should probably happen in the next couple of months.

godmiaozi, 2020-07-07:
@dbatyai thanks for your reply. i still do not understand the following description"
` Using 4 byte alignment would've also meant that we would have to align memory manually after allocation for values that require it (like doubles), which would've wasted additional memory and potentially caused further fragmentation.`
if we use 4 bytes alignment， double will take 8bytes，why will we have more fragment?

akosthekiss, 2020-07-07:
Because on several systems, a `double` has to be aligned to 8 bytes. So, if memory chunks were 4-bytes-aligned, we'd have to deal with alignment of `double`s manually. I.e., we'd have to allocate 12 bytes to guarantee that there are 8 consecutive bytes in the chunk that start on an 8-bytes-aligned address.

Try `alignof.c`:
```c
#include <stdalign.h>
#include <stdio.h>

int main() {
    printf("align double to %d bytes\n", alignof(double));
    return 0;
}
```

Compile and run:
```
$ gcc -o alignof alignof.c 
$ ./alignof 
align double to 8 bytes
```

Note that `alignof()` is C11, not C99, but that shouldn't matter in this demonstration.

Also note that alignment of `double`s is implementation-dependent in theory, but is quite often 8 bytes in practice.

rerobika, 2020-07-13:
I think we can treat this question answered.

closed 2020-07-13 by rerobika