jerryscript-project/jerryscript #3994: Question about this_val of jerry_call_function

weixiongmei, 2020-07-10:
Hi, I'm having hard time for getting the correct "this" context from the native callback by calling jerry_call_function, it always returned "global".

```javascript
export class ClassA
{
       print()
      {
		console.log("This is from Class A\n");
      }
	set cb_event(cb)
	{
		native_driver_timer.setTimeoutEvent(cb);
	}
	constructor(){}
}
```

```c
static jerry_value_t function_native_driver_timer_setTimeoutEvent(const jerry_value_t func_value, /**< function object */
                                 const jerry_value_t this_value, /**< this arg */
                                 const jerry_value_t args[],    /**< function arguments */
                                 const jerry_length_t args_cnt)  /**< number of function arguments */
{
  jerry_release_value(jerry_call_function(args[0],
                                            this_val,
                                            NULL,
                                            0));
  return jerry_create_undefined();
}
```

```javascript
let ca = new ClassA();
ca.cb_event = (() => {
	this.print(); // Undefined.
	console.log(JSON.stringify(this)); //Always returning "global", wondering how to make "this" as a context of variable "ca".
});
```

galpeter, 2020-07-10:
Hi,

Given the following example JS code from your report:

```js
let ca = new ClassA();
ca.cb_event = (() => {
	this.print(); // Undefined.
	console.log(JSON.stringify(this)); //Always returning "global", wondering how to make "this" as a context of variable "ca".
});
```

The arrow function's `this` value is already bound to the `this` value where the arrow function was created. In this case the global object. There is a good post on how arrow function's `this` value works: https://www.codementor.io/@dariogarciamoya/understanding-this-in-javascript-with-arrow-functions-gcpjwfyuc

Try using a normal function instead of an arrow function to see that the `this` value is the one you want.

rerobika, 2020-07-27:
I think the question has been answered so we can treat this issue resolved.

closed 2020-07-27 by rerobika