jerryscript-project/jerryscript #4057: Please am new here, i want to know more about coding!

aoegold, 2020-07-25:
please i don't really know how to convert and edit this code below, and please i really want to know what is inside the script
<script type='text/javascript'>
//<![CDATA[
var _0xb2ac=["\x36\x5A\x28\x36\x52\x28\x70\x2C\x61\x2C\x63\x2C\x6B\x2C\x65\x2C\x72\x29\x7B\x65\x3D\x36\x52\x28\x63\x29\x7B\x36\x53\x28\x63\x3C\x61\x3F\x27\x27\x3A\x65\x28\x36\x55\x28\x63\x2F\x61\x29\x29\x29\x2B\x28\x28\x63\x3D\x63\x25\x61\x29\x3E\x33\x35\x3F\x36\x56\x2E\x37\x30\x28\x63\x2B\x32\x39\x29\x3A\x63\x2E\x37\x31\x28\x33\x36\x29\x29\x7D\x3B\x36\x54\x28\x21\x27\x27\x2E\x36\x57\x28\x2F\x5E\x2F\x2C\x36\x56\x29\x29\x7B\x36\x58\x28\x63\x2D\x2D\x29\x72\x5B\x65\x28\x63\x29\x5D\x3D\x6B\x5B\x63\x5D\x7C\x7C\x65\x28\x63\x29\x3B\x6B\x3D\x5B\x36\x52\x28\x65\x29\x7B\x36\x53\x20\x72\x5B\x65\x5D\x7D\x5D\x3B\x65\x3D\x36\x52\x28\x29\x7B\x36\x53\x27\x5C\x5C\x77\x2B\x27\x7D\x3B\x63\x3D\x31\x7D\x3B\x36\x58\x28\x63\x2D\x2D\x29\x36\x54\x28\x6B\x5B\x63\x5D\x29\x70\x3D\x70\x2E\x36\x57\x28\x36\x59\x20\x37\x32\x28\x27\x5C\x5C\x62\x27\x2B\x65\x28\x63\x29\x2B\x27\x5C\x5C\x62\x27\x2C\x27\x67\x27\x29\x2C\x6B\x5B\x63\x5D\x29\x3B\x36\x53\x20\x70\x7D\x28\x27\x45\x20\x33\x6F\x3D\x5B\x5C\x27\x5C\x5C\x35\x5C\x5C\x6E\x5C\x5C\x6E\x5C\x5C\x63\x5C\x5C\x4C\x5C\x27\x2C\x5C\x27\x5C\x5C\x76\x5C\x5C\x66\x5C\x5C\x38\x5C\x5C\x33\x5C\x5C\x72\x5C\x5C\x39\x5C\x5C\x66\x5C\x5C\x33\x5C\x5C\x38\x5C\x27\x2C\x5C\x27\x5C\x5C\x31\x37\x5C\x5C\x33\x5C\x5C\x41\x5C\x5C\x31\x6B\x5C\x5C\x31\x5C\x5C\x6E\x5C\x27\x2C\x5C\x27\x5C\x5C\x32\x66\x5C\x5C\x33\x70\x5C\x5C\x32\x44\x5C\x5C\x32\x66\x5C\x5C\x6F\x5C\x5C\x32\x45\x5C\x5C\x32\x32\x5C\x5C\x33\x70\x5C\x5C\x6F\x5C\x5C\x32\x32\x5C\x27\x2C\x5C\x27\x5C\x5C\x32\x44\x5C\x5C\x32\x66\x5C\x5C\x6F\x5C\x5C\x32\x45\x5C\x5C\x32\x32\x5C\x5C\x33\x71\x5C\x5C\x32\x32\x5C\x5C\x33\x71\x5C\x5C\x32\x32\x5C\x5C\x32\x44\x5C\x27\x2C\x5C\x27\x5C\x5C\x32\x66\x5C\x5C\x6F\x5C\x5C\x32\x45\x5C\x5C\x31\x6C\x5C\x27\x2C\x5C\x27\x5C\x5C\x56\x5C\x5C\x6C\x5C\x5C\x4C\x5C\x5C\x61\x5C\x5C\x64\x5C\x5C\x66\x5C\x5C\x36\x5C\x5C\x33\x5C\x5C\x66\x5C\x5C\x36\x5C\x27\x2C\x5C\x27\x5C\x5C\x50\x5C\x5C\x48\x5C\x5C\x39\x5C\x5C\x37\x5C\x5C\x39\x5C\x5C\x68\x5C\x5C\x63\x5C\x5C\x33\x5C\x27\x2C\x5C\x27\x5C\x5C\x63\x5C\x5C\x64\x5C\x5C\x61\x5C\x5C\x35\x5C\x5C\x36\x5C\x5C\x39\x5C\x5C\x64\x5C\x5C\x66\x5C\x27\x2C\x5C\x27\x5C\x5C\x74\x5C\x5C\x65\x5C\x5C\x33\x5C\x5C\x72\x5C\x27\x2C\x5C\x27\x5C\x5C\x42\x5C\x5C\x36\x5C\x5C\x33\x5C\x5C\x6C\x5C\x5C\x6E\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x36\x5C\x5C\x33\x5C\x5C\x37\x5C\x27\x2C\x5C\x27\x5C\x5C\x4C\x5C\x5C\x35\x5C\x5C\x65\x5C\x5C\x38\x5C\x5C\x42\x5C\x5C\x61\x5C\x5C\x64\x5C\x5C\x6C\x5C\x5C\x78\x5C\x27\x2C\x5C\x27\x5C\x5C\x64\x5C\x5C\x66\x5C\x5C\x63\x5C\x5C\x64\x5C\x5C\x35\x5C\x5C\x38\x5C\x27\x2C\x5C\x27\x5C\x5C\x41\x5C\x5C\x33\x5C\x5C\x36\x5C\x5C\x31\x6B\x5C\x5C\x63\x5C\x5C\x33\x5C\x5C\x6C\x5C\x5C\x33\x5C\x5C\x66\x5C\x5C\x36\x5C\x27\x2C\x5C\x27\x5C\x5C\x31\x34\x5C\x5C\x4C\x5C\x5C\x31\x30\x5C\x5C\x38\x5C\x27\x2C\x5C\x27\x5C\x5C\x6C\x5C\x5C\x4C\x5C\x5C\x61\x5C\x5C\x64\x5C\x5C\x66\x5C\x5C\x36\x5C\x5C\x33\x5C\x5C\x66\x5C\x5C\x36\x5C\x27\x2C\x5C\x27\x5C\x5C\x37\x5C\x5C\x33\x5C\x5C\x36\x5C\x5C\x51\x5C\x5C\x36\x5C\x5C\x36\x5C\x5C\x65\x5C\x5C\x39\x5C\x5C\x68\x5C\x5C\x76\x5C\x27\x2C\x5C\x27\x5C\x5C\x74\x5C\x5C\x36\x5C\x5C\x36\x5C\x5C\x6E\x5C\x5C\x50\x5C\x5C\x78\x5C\x5C\x78\x5C\x5C\x4A\x5C\x5C\x4A\x5C\x5C\x4A\x5C\x27\x2C\x5C\x27\x5C\x5C\x38\x5C\x5C\x64\x5C\x5C\x72\x5C\x5C\x64\x5C\x5C\x63\x5C\x5C\x63\x5C\x5C\x64\x5C\x5C\x4A\x5C\x27\x2C\x5C\x27\x5C\x5C\x36\x5C\x5C\x39\x5C\x5C\x36\x5C\x5C\x63\x5C\x5C\x33\x5C\x27\x2C\x5C\x27\x5C\x5C\x31\x38\x5C\x5C\x65\x5C\x5C\x33\x5C\x5C\x33\x5C\x5C\x6F\x5C\x5C\x54\x5C\x5C\x74\x5C\x5C\x33\x5C\x5C\x6C\x5C\x5C\x33\x5C\x27\x2C\x5C\x27\x5C\x5C\x38\x5C\x5C\x39\x5C\x5C\x37\x5C\x5C\x6E\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x4C\x5C\x5C\x50\x5C\x5C\x6F\x5C\x5C\x39\x5C\x27\x2C\x5C\x27\x5C\x5C\x36\x5C\x5C\x31\x35\x5C\x5C\x6F\x5C\x5C\x72\x5C\x5C\x64\x5C\x5C\x66\x5C\x5C\x36\x5C\x5C\x6A\x5C\x5C\x37\x5C\x5C\x39\x5C\x27\x2C\x5C\x27\x5C\x5C\x36\x5C\x5C\x31\x6D\x5C\x5C\x39\x5C\x5C\x6C\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x65\x5C\x5C\x36\x5C\x5C\x35\x5C\x5C\x66\x5C\x27\x2C\x5C\x27\x5C\x5C\x36\x5C\x5C\x31\x35\x5C\x5C\x6F\x5C\x5C\x61\x5C\x5C\x64\x5C\x5C\x63\x5C\x5C\x64\x5C\x5C\x65\x5C\x5C\x50\x5C\x5C\x6F\x5C\x27\x2C\x5C\x27\x5C\x5C\x56\x5C\x5C\x30\x5C\x5C\x68\x5C\x5C\x33\x5C\x5C\x73\x5C\x5C\x35\x5C\x5C\x72\x5C\x5C\x31\x6D\x5C\x5C\x39\x5C\x5C\x6C\x5C\x27\x2C\x5C\x27\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x65\x5C\x5C\x36\x5C\x5C\x35\x5C\x5C\x66\x5C\x5C\x36\x5C\x5C\x31\x35\x5C\x5C\x6F\x5C\x5C\x48\x5C\x27\x2C\x5C\x27\x5C\x5C\x6A\x5C\x5C\x39\x5C\x5C\x66\x5C\x5C\x38\x5C\x5C\x33\x5C\x5C\x31\x5C\x5C\x50\x5C\x5C\x7A\x5C\x5C\x7A\x5C\x5C\x31\x6D\x5C\x27\x2C\x5C\x27\x5C\x5C\x6F\x5C\x5C\x64\x5C\x5C\x6E\x5C\x5C\x35\x5C\x5C\x61\x5C\x5C\x39\x5C\x5C\x36\x5C\x5C\x4C\x5C\x5C\x50\x5C\x5C\x6F\x5C\x27\x2C\x5C\x27\x5C\x5C\x34\x5C\x5C\x31\x6D\x5C\x5C\x39\x5C\x5C\x6C\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x65\x5C\x5C\x36\x5C\x5C\x35\x5C\x5C\x66\x5C\x27\x2C\x5C\x27\x5C\x5C\x36\x5C\x5C\x31\x35\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x37\x5C\x5C\x39\x5C\x5C\x36\x5C\x5C\x39\x5C\x5C\x64\x5C\x5C\x66\x5C\x27\x2C\x5C\x27\x5C\x5C\x39\x5C\x5C\x6C\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x65\x5C\x5C\x36\x5C\x5C\x35\x5C\x5C\x66\x5C\x5C\x36\x5C\x5C\x31\x35\x5C\x27\x2C\x5C\x27\x5C\x5C\x39\x5C\x5C\x66\x5C\x5C\x66\x5C\x5C\x33\x5C\x5C\x65\x5C\x5C\x31\x41\x5C\x5C\x54\x5C\x5C\x32\x33\x5C\x5C\x59\x5C\x27\x2C\x5C\x27\x5C\x5C\x54\x5C\x5C\x33\x5C\x5C\x6C\x5C\x5C\x6E\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x36\x5C\x5C\x33\x5C\x5C\x37\x5C\x5C\x31\x74\x5C\x27\x2C\x5C\x27\x5C\x5C\x35\x5C\x5C\x65\x5C\x5C\x38\x5C\x27\x2C\x5C\x27\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x31\x36\x5C\x5C\x4C\x5C\x5C\x4C\x5C\x5C\x35\x5C\x5C\x65\x5C\x5C\x38\x5C\x27\x2C\x5C\x27\x5C\x5C\x33\x5C\x5C\x35\x5C\x5C\x61\x5C\x5C\x74\x5C\x27\x2C\x5C\x27\x5C\x5C\x35\x5C\x5C\x36\x5C\x5C\x36\x5C\x5C\x65\x5C\x27\x2C\x5C\x27\x5C\x5C\x37\x5C\x5C\x65\x5C\x5C\x61\x5C\x27\x2C\x5C\x27\x5C\x5C\x65\x5C\x5C\x64\x5C\x5C\x76\x5C\x5C\x66\x5C\x5C\x38\x5C\x27\x2C\x5C\x27\x5C\x5C\x4A\x5C\x5C\x39\x5C\x5C\x38\x5C\x5C\x36\x5C\x5C\x74\x5C\x27\x2C\x5C\x27\x5C\x5C\x74\x5C\x5C\x33\x5C\x5C\x39\x5C\x5C\x41\x5C\x5C\x74\x5C\x5C\x36\x5C\x27\x2C\x5C\x27\x5C\x5C\x6A\x5C\x5C\x6E\x5C\x5C\x6A\x5C\x5C\x4D\x5C\x5C\x6A\x5C\x5C\x66\x5C\x5C\x64\x5C\x5C\x6A\x5C\x5C\x66\x5C\x5C\x76\x5C\x27\x2C\x5C\x27\x5C\x5C\x6C\x5C\x5C\x35\x5C\x5C\x36\x5C\x5C\x61\x5C\x5C\x74\x5C\x27\x2C\x5C\x27\x5C\x5C\x37\x5C\x5C\x70\x5C\x5C\x6B\x5C\x5C\x6A\x5C\x5C\x61\x5C\x27\x2C\x5C\x27\x5C\x5C\x65\x5C\x5C\x33\x5C\x5C\x6E\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x61\x5C\x5C\x33\x5C\x27\x2C\x5C\x27\x5C\x5C\x78\x5C\x5C\x37\x5C\x5C\x70\x5C\x5C\x6B\x5C\x5C\x6A\x5C\x5C\x61\x5C\x27\x2C\x5C\x27\x5C\x5C\x4A\x5C\x5C\x70\x5C\x5C\x6B\x5C\x5C\x6A\x5C\x5C\x74\x5C\x27\x2C\x5C\x27\x5C\x5C\x78\x5C\x5C\x4A\x5C\x5C\x70\x5C\x5C\x6B\x5C\x5C\x6A\x5C\x5C\x74\x5C\x5C\x70\x5C\x5C\x6B\x5C\x5C\x6A\x5C\x5C\x6E\x5C\x27\x2C\x5C\x27\x5C\x5C\x6A\x5C\x5C\x4D\x5C\x5C\x6A\x5C\x5C\x66\x5C\x5C\x64\x5C\x5C\x6A\x5C\x5C\x66\x5C\x5C\x76\x5C\x27\x2C\x5C\x27\x5C\x5C\x63\x5C\x5C\x64\x5C\x5C\x35\x5C\x5C\x38\x5C\x5C\x6F\x5C\x5C\x65\x5C\x5C\x33\x5C\x5C\x37\x5C\x5C\x39\x5C\x5C\x31\x36\x5C\x27\x2C\x5C\x27\x5C\x5C\x37\x5C\x5C\x61\x5C\x5C\x65\x5C\x5C\x64\x5C\x5C\x63\x5C\x5C\x63\x5C\x5C\x54\x5C\x5C\x64\x5C\x5C\x6E\x5C\x27\x2C\x5C\x27\x5C\x5C\x64\x5C\x5C\x72\x5C\x5C\x72\x5C\x5C\x37\x5C\x5C\x33\x5C\x5C\x36\x5C\x27\x2C\x5C\x27\x5C\x5C\x36\x5C\x5C\x64\x5C\x5C\x6E\x5C\x27\x2C\x5C\x27\x5C\x5C\x35\x5C\x5C\x38\x5C\x5C\x38\x5C\x5C\x53\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x37\x5C\x5C\x37\x5C\x27\x2C\x5C\x27\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x31\x36\x5C\x5C\x4C\x5C\x5C\x6A\x5C\x5C\x4C\x5C\x5C\x35\x5C\x5C\x65\x5C\x5C\x38\x5C\x27\x2C\x5C\x27\x5C\x5C\x42\x5C\x5C\x39\x5C\x5C\x66\x5C\x5C\x38\x5C\x5C\x33\x5C\x5C\x31\x5C\x5C\x6A\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x37\x5C\x27\x2C\x5C\x27\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x37\x5C\x5C\x36\x5C\x5C\x6A\x5C\x5C\x36\x5C\x5C\x74\x5C\x5C\x76\x5C\x5C\x6C\x5C\x5C\x68\x5C\x27\x2C\x5C\x27\x5C\x5C\x31\x31\x5C\x5C\x6F\x5C\x5C\x42\x5C\x5C\x31\x6E\x5C\x5C\x64\x5C\x5C\x6E\x5C\x5C\x76\x5C\x5C\x63\x5C\x5C\x35\x5C\x5C\x65\x5C\x27\x2C\x5C\x27\x5C\x5C\x31\x6E\x5C\x5C\x64\x5C\x5C\x37\x5C\x5C\x36\x5C\x5C\x37\x5C\x5C\x6F\x5C\x5C\x42\x5C\x5C\x6E\x5C\x5C\x64\x5C\x5C\x37\x5C\x27\x2C\x5C\x27\x5C\x5C\x36\x5C\x5C\x6A\x5C\x5C\x39\x5C\x5C\x6C\x5C\x5C\x35\x5C\x5C\x41\x5C\x5C\x33\x5C\x5C\x6A\x5C\x5C\x63\x5C\x5C\x39\x5C\x27\x2C\x5C\x27\x5C\x5C\x74\x5C\x5C\x76\x5C\x5C\x6C\x5C\x5C\x68\x5C\x5C\x31\x31\x5C\x5C\x6F\x5C\x5C\x42\x5C\x5C\x31\x38\x5C\x5C\x33\x5C\x5C\x35\x5C\x27\x2C\x5C\x27\x5C\x5C\x4 [... cut ...] \x32\x70\x29\x29\x7D\x7D\x4B\x28\x43\x28\x29\x7B\x45\x20\x32\x64\x3D\x31\x70\x5B\x32\x28\x5C\x27\x5C\x5C\x30\x5C\x5C\x31\x5C\x5C\x34\x5C\x5C\x73\x5C\x5C\x7A\x5C\x27\x29\x2B\x32\x28\x5C\x27\x5C\x5C\x30\x5C\x5C\x31\x5C\x5C\x34\x5C\x5C\x73\x5C\x5C\x35\x5C\x27\x29\x5D\x28\x5C\x27\x5C\x5C\x37\x5C\x5C\x61\x5C\x5C\x65\x5C\x5C\x39\x5C\x5C\x6E\x5C\x5C\x36\x5C\x27\x29\x3B\x32\x64\x5B\x32\x28\x5C\x27\x5C\x5C\x30\x5C\x5C\x31\x5C\x5C\x34\x5C\x5C\x73\x5C\x5C\x61\x5C\x27\x29\x5D\x3D\x32\x28\x5C\x27\x5C\x5C\x30

akosthekiss, 2020-07-25:
@aoegold How is this related to the JerryScript project?

aoegold, 2020-07-25:
am sorry @akosthekiss i dont where to post, but i have said it before that am new here

akosthekiss, 2020-07-25:
@aoegold I'm not sure what "here" means in this context. Are you new to the JerryScript project, are you new to the JavaScript language, are you new to the GitHub platform, are you new to programming, or ...? If your question is not related to the project, I'd suggest you post it to the appropriate platform (e.g., StackOverflow).

rerobika, 2020-07-27:
This question is out of the project's scope. Please use the mentioned platforms suggested by @akosthekiss .

closed 2020-07-27 by rerobika