jerryscript-project/jerryscript #4058: Question: How to handle Date object in native C?

weixiongmei, 2020-07-26:
Hi, I'm having hard time to figure out how to get the Date.year, month, date in the native C code. Also having hard time to figure out how to return a Date object from native to javascript. Thanks

Javascript
```javascript
let date = native_driver_gui_calendar.setTodayDate(new Date(2020, 6, 10));
```

Native C
```C
static jerry_value_t function_native_driver_gui_calendar_setTodayDate(const jerry_value_t func_value, /**< function object */
                                 const jerry_value_t this_value, /**< this arg */
                                 const jerry_value_t args[],    /**< function arguments */
                                 const jerry_length_t args_cnt)  /**< number of function arguments */
{
  //How to get the year, month, date of a Date object?
  //The following code returns all zero
  printf("Year:%d Month:%d Day:%d\n", ecma_date_year_from_time(args[0]), ecma_date_month_from_time(args[0]), ecma_date_date_from_time(args[0]));

 //How to return "new Date(2020, 0,0)" value to the javascript?
  return jerry_create_date();
}
```

rerobika, 2020-07-27:
Hello @weixiongmei !

Unfortunately, currently we do not have any API functions to handle date objects.
Also the reason is why the the called `ecma` level functions are not working, it that these methods expect an `ecma_number_t` instead of a `jerry_value_t`. If you unpack it into a double variable with `jerry_get_number_from_value(args[0])` you can use these internal methods, with the warning of the usage of non-api functions is not the best practice.

Moreover I can suggest you two __better__ solutions:
**A:**
 - ecma_date_year_from_time(args[0]) is almost equivalent to perform `Date.prototype.getYear.call(args[0])`
 - `.call()` can be performed with `jerry_invoke_function` with `args[0]` as `this` argument 
 - `getYear` is the property of `Date.prototype`
 - `Date.prototype` is the property of `Date`
 - `Date` is the property of the `global object.`
 - So putting the pieces together,  use `jerry_get_property (global, "Date")`, then `jerry_get_property(Date, "prototype")`, then `jerry_get_property(Date.prototype, "getYear")` and finally `jerry_invoke_function (Date.prototype.getYear, args[0], NULL, 0)`

**B:**
 - If you have free-time/capacity it'd be great to publish these internal method to the API. I think we should have one single universal function for retrieving information form a date object. So my proposal would look like this:

```c
typedef enum
{
  JERRY_DATE_OPERATION_GET_YEAR,
  JERRY_DATE_OPERATION_GET_MONTH,
  // ... and so on
} jerry_date_operation_t;

jerry_value_t
jerry_date_get (jerry_date_operation_t operation, jerry_value_t value)
{
  // step 1: validate date object
  
  // step 2: get the internal date
  ecma_number_t date_num = get_the_stored_internal_date_value() // see ecma_builtin_date_prototype_dispatch_routine
  // step 3: choose the operation
  switch (operation) {
    case JERRY_DATE_OPERATION_GET_YEAR: {
      return ecma_date_year_from_time (date_num)
    }
    case JERRY_DATE_OPERATION_GET_MONTH: {
      return ecma_date_month_from_time (date_num)
    }

   // ... and so on
  }
}
```

Both options are suitable for resolve your problem, but IMHO the option **B** would be more future-proof also much efficient as well.

weixiongmei, 2020-07-29:
@rerobika Thank you so much~~

closed 2020-08-03 by rerobika