jerryscript-project/jerryscript #4135: Assertion CBC_FUNCTION_GET_TYPE (frame_ctx_p->bytecode_header_p->status_flags) == CBC_FUNCTION_ASYNC in opfunc_async_create_and_await

ossy-szeged, 2020-08-12:
###### JerryScript revision
0c154306a87492f255edae6f483a768f128c2c09 
(latest master - 2020.08.11.)

###### Build platform
Ubuntu 18.04.5 LTS (Linux 4.15.0-112-generic x86_64)

###### Build steps
tools/build.py --debug --logging=on --error-messages=on --line-info=on

###### Test case
found by test262-esnext language/module-code/top-level-await/... tests

minimized test case
```
function fn() { return function() {}; }
export class C extends fn(await []) {};
```

###### Output
```
ICE: Assertion 'CBC_FUNCTION_GET_TYPE (frame_ctx_p->bytecode_header_p->status_flags) == CBC_FUNCTION_ASYNC || CBC_FUNCTION_GET_TYPE (frame_ctx_p->bytecode_header_p->status_flags) == CBC_FUNCTION_ASYNC_ARROW' failed at /home/oszi/jerryscript/jerry-core/vm/opcodes.c(opfunc_async_create_and_await):914.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted (core dumped)
```

###### Expected behavior
not assert, but SyntaxError, because top-level-await isn't supported yet

ossy-szeged, 2020-09-29:
Still valid on latest master ( 0ffe1665bd0ee457dd4c1083811e1592a5e2908d ).

zherczeg, 2020-12-11:
I get a SyntaxError now, maybe we can close this?

rerobika, 2020-12-11:
Closed via #4266 .

closed 2020-12-11 by rerobika