jerryscript-project/jerryscript #4157: No class support

szdytom, 2020-08-19:
when I input `class A {}`, jerry-cli returns `Script Error: SyntaxError`
(can't work as `var A = class {}` too)

Linux-5.4.0-42-generic-x86_64-with-glibc2.29

akosthekiss, 2020-08-19:
@szdytom How did you build the project (and which version)? In previous versions, es5.1 was default and the so-called es2015-subset profile had to be selected explicitly. Latest master has es.next profile as the default, though.

szdytom, 2020-08-19:
well, I downloaded the Latest release(JerryScript 2.3.0 on 12 Jun), unzipped it and ran `python3 tools/build.py`

rerobika, 2020-08-19:
The default profile in the `2.3` release was the `es5.1` therefore if you need to pass the `profile=es2015-subset` for `tools/build.py` if you insist to this version or use the latest master where the default profile is `es.next` as @akosthekiss suggested.

akosthekiss, 2020-08-19:
There was a significant change in the default language version, see #3737 . Version 2.3.0 was the last release to have ES5.1 as default. If you plan to use that version, you will have to add `--profile=es2015-subset` to your build command line to enable classes. If you are OK with using the master branch, then ES.Next will already be the default profile, i.e., classes are enabled.

szdytom, 2020-08-19:
Problem solved! thanks!

closed 2020-08-19 by szdytom