jerryscript-project/jerryscript #4167: Slow Start Up For Large Project

weixiongmei, 2020-08-26:
My project has over a 10000 lines of javascript codes, it takes almost 13 seconds on ESP32(240Mhz) to load the codes from the internal flash, the snapshot will be very helpful for large projects, but unfortunately the modules doesn't support for snapshot, is it possible to add snapshot support for modules? Thanks

philippe44, 2020-09-07:
I'm just starting with JerryScript and esp32. Where are you with your port? Is there a known reason why snapshot don't work? Have you thought about contributing back your port to the project?

zherczeg, 2020-09-07:
The snapshot support for modules is not a trivial task, since the import/export support, which is normally done by the parser should be done at snapshot loading time. Unfortunately we don't have resources to work on it at the moment.

philippe44, 2020-09-07:
understood - thanks for the answer. BTW, do you want an esp32 port or that does not matter for you?

weixiongmei, 2020-09-07:
@philippe44, sorry, the port is for commercial usage, so can't share it, spent two years to port all the drivers and GUI to it.

philippe44, 2020-09-07:
no worries - I was of course just thinking about the base jerry-port things, not more. I'll share mine if the maintainers are interested

dbatyai, 2020-09-08:
@philippe44, if you're willing to implement it, then sure. It would be good to add a build job for it to the CI.