jerryscript-project/jerryscript #4183: Using 32 bits float and -Wconversion

philippe44, 2020-09-06:
I'm a bit puzzled as what happens when using 32 bits float (-DJERRY_NUMBER_TYPE_FLOAT64=0) in various places like line 350 of ecma-helpers-conversion.c (and few other places). By default, when using GCC (I'm portin to the esp32 platform, so it's gcc), the root CMakeLists.txt set "conversion" as a warning and an error but obviously, 
```
num = num * radix + (ecma_number_t) digit_value;
```
Produces an error as we are trying to represent a 32 bits int into a 4 bytes float, so we loose precision. I surely can change CMakeLists.txt or the code itself, but is there any preferred way before I submit my port?

ossy-szeged, 2020-09-07:
In this case it's safe to cast radix to ecma_number_t, because it can be only 2, 8 or 16,
Unfortunately it seems there are other build failures related to -DJERRY_NUMBER_TYPE_FLOAT64=0.

dbatyai, 2020-09-08:
Preferably these should be fixed in the code itself, and only disable warnings if absolutely necessary.

philippe44, 2020-09-11:
ok, done in the code, was just a few casts

closed 2020-09-17 by philippe44