jerryscript-project/jerryscript #4188: Invalid memory access (segfault) in scanner_seek in js-scanner-util.c

ruc-fuzzer, 2020-09-10:
###### JerryScript revision
```
The master branch or git hash: c013fad
```

###### Build platform
```
Ubuntu 18.04.4 LTS  Linux 4.15.0-74-generic x86_64
```

###### Build steps
```
Build with ASAN enabled:
python tools/build.py  --compile-flag=-fsanitize=address --compile-flag=-g --strip=OFF
```
###### Test case
```
 { ;  let a = a } ;  export { } ;  for ( let in 
```

###### Output
```
ASAN:DEADLYSIGNAL
=================================================================
==18474==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000008 (pc 0x5555555f928e bp 0x7fffffffddb4 sp 0x7fffffffcf30 T0)
==18474==The signal is caused by a READ memory access.
==18474==Hint: address points to the zero page.
    #0 0x5555555f928d in scanner_seek /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-scanner-util.c:370
    #1 0x5555555d5981 in parser_parse_for_statement_start /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:1269
    #2 0x5555555d5981 in parser_parse_statements /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:2967
    #3 0x555555605563 in parser_parse_source /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser.c:2304
    #4 0x555555605e83 in parser_parse_script /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser.c:2933
    #5 0x555555613d4e in jerry_parse.constprop.28 /home/user/Product/jerryscript/master/src/jerry-core/api/jerry.c:451
    #6 0x55555556c3ea in main /home/user/Product/jerryscript/master/src/jerry-main/main-unix.c:112
    #7 0x7ffff6a4ab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #8 0x55555556d5d9 in _start (/home/user/Product/jerryscript/master/src/build/bin/jerry+0x195d9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-scanner-util.c:370 in scanner_seek
==18474==ABORTING
```

ossy-szeged, 2020-09-29:
Still valid on latest master ( 0ffe1665bd0ee457dd4c1083811e1592a5e2908d ).
It is an internal assertion:
ICE: Assertion '!has_context || context_p->next_scanner_info_p->type == SCANNER_TYPE_BLOCK' failed at /home/oszi/jerryscript/jerry-core/parser/js/js-parser-statm.c(parser_parse_for_statement_start):1225.
Error: ERR_FAILED_INTERNAL_ASSERTION

rerobika, 2021-01-18:
Resolved via #4462.

closed 2021-01-18 by rerobika