jerryscript-project/jerryscript #4190: heap buffer overflow in lexer_convert_ident_to_cesu8 in js-lexer.c

ruc-fuzzer, 2020-09-11:
##### JerryScript revision
```
The master branch or git hash: c013fad
```

###### Build platform
```
Ubuntu 18.04.4 LTS  Linux 4.15.0-74-generic x86_64
```

###### Build steps
```
Build with ASAN enabled:
python tools/build.py  --compile-flag=-fsanitize=address --compile-flag=-g --strip=OFF
```

###### Test case
```
 for ( a in a ) ;  'a' ; 'a' ;  + eval `a` ;  a = 1 ;  for ( a in a ) a ;  a = a => a ;  a ++ ;  a . a `a` ;  export default a ;  'a' ; 'a' ;  + eval `a` ;  a = 1 ;  for ( a in a ) a ;  for ( let )
```

###### Output
```
==6939==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x61100000011e at pc 0x5555555c2e91 bp 0x7fffffffcb30 sp 0x7fffffffcb20
READ of size 1 at 0x61100000011e thread T0
    #0 0x5555555c2e90 in lexer_convert_ident_to_cesu8 /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-lexer.c:2146
    #1 0x5555555c34ca in lexer_convert_literal_to_chars /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-lexer.c:2200
    #2 0x5555555c3bbd in lexer_construct_literal_object /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-lexer.c:2422
    #3 0x5555555fe4d0 in scanner_create_variables /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-scanner-util.c:2171
    #4 0x5555555c1184 in parser_push_block_context /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:367
    #5 0x5555555d6746 in parser_parse_for_statement_start /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:1533
    #6 0x5555555d6746 in parser_parse_statements /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:2967
    #7 0x555555605563 in parser_parse_source /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser.c:2304
    #8 0x555555605e83 in parser_parse_script /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser.c:2933
    #9 0x555555613d4e in jerry_parse.constprop.28 /home/user/Product/jerryscript/master/src/jerry-core/api/jerry.c:451
    #10 0x55555556c3ea in main /home/user/Product/jerryscript/master/src/jerry-main/main-unix.c:112
    #11 0x7ffff6a4ab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #12 0x55555556d5d9 in _start (/home/user/Product/jerryscript/master/src/build/bin/jerry+0x195d9)
```

ossy-szeged, 2020-09-29:
Still valid on latest master ( 0ffe1665bd0ee457dd4c1083811e1592a5e2908d ).
ICE: Assertion 'context_p->next_scanner_info_p->type == SCANNER_TYPE_BLOCK' failed at /home/oszi/jerryscript/jerry-core/parser/js/js-parser-statm.c(parser_push_block_context):344.

rerobika, 2021-01-18:
Resolved via #4462.

closed 2021-01-18 by rerobika