jerryscript-project/jerryscript #4195: heap-buffer-overflow in lexer_construct_literal_object in js-lexer.c

ruc-fuzzer, 2020-09-14:
###### JerryScript revision
The master branch or git hash: c013fad

###### Build platform
Ubuntu 18.04.4 LTS  Linux 4.15.0-74-generic x86_64

###### Build steps
Build with ASAN enabled:
python tools/build.py  --compile-flag=-fsanitize=address --compile-flag=-g

###### Test case
a => AFL ;  let a ;  ! class extends a { } ;  a = a => a ;  a ++ ;  a . a `a` ;  export default a ;  'a' ; 'a' ;  + eval `a` ;  a = 1 ;  for ( a in a ) a ;  for ( let 

###### Output
==3997==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60f0000000e8 at pc 0x7ffff6ec9bb5 bp 0x7fffffffcd50 sp 0x7fffffffc4f8
READ of size 4 at 0x60f0000000e8 thread T0
    #0 0x7ffff6ec9bb4  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xafbb4)
    #1 0x555555602095 in lexer_construct_literal_object /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-lexer.c:2450
    #2 0x5555555c0477 in scanner_create_variables /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-scanner-util.c:2171
    #3 0x55555560edab in parser_push_block_context /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:367
    #4 0x555555613121 in parser_parse_for_statement_start /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:1533
    #5 0x555555613121 in parser_parse_statements /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser-statm.c:2967
    #6 0x5555555ba8c5 in parser_parse_source /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser.c:2304
    #7 0x5555555bc12d in parser_parse_script /home/user/Product/jerryscript/master/src/jerry-core/parser/js/js-parser.c:2933
    #8 0x555555573b39 in jerry_parse /home/user/Product/jerryscript/master/src/jerry-core/api/jerry.c:451
    #9 0x55555556e42b in main /home/user/Product/jerryscript/master/src/jerry-main/main-unix.c:112
    #10 0x7ffff6a4ab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #11 0x555555570f69 in _start (/home/user/Product/jerryscript/master/src/build/bin/jerry+0x1cf69)

0x60f0000000e8 is located 0 bytes to the right of 168-byte region [0x60f000000040,0x60f0000000e8)
allocated by thread T0 here:
    #0 0x7ffff6ef8b40 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb40)
    #1 0x55555561a80f in jerry_port_read_source /home/user/Product/jerryscript/master/src/jerry-port/default/default-module.c:58
    #2 0x7ffff6a4ab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: heap-buffer-overflow (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xafbb4) 
Shadow bytes around the buggy address:
  0x0c1e7fff7fc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c1e7fff7fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c1e7fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c1e7fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c1e7fff8000: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
=>0x0c1e7fff8010: 00 00 00 00 00 00 00 00 00 00 00 00 00[fa]fa fa
  0x0c1e7fff8020: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7fff8030: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7fff8060: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==3997==ABORTING

ZhangZhuoSJTU, 2020-09-15:
Reproducible. I have tested this on the newest commit, and the overflow still exists. 

If I simplify the PoC as following.
```js
a => AFL ; 
a . a `a` ;
export default a ; 
+ eval `a` ; 
for ( a in a ) a ;
for ( let
```
I will get following report from ASAN.
```base
=================================================================
==26638==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x608000000075 at pc 0x7fe12c823bf5 bp 0x7ffdaa90ea50 sp 0x7ffdaa90e1f8
READ of size 4 at 0x608000000075 thread T0
    #0 0x7fe12c823bf4  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xafbf4)
    #1 0x4850b1 in lexer_construct_literal_object /home/XXX/git/jerryscript/jerry-core/parser/js/js-lexer.c:2450
    #2 0x485cb4 in scanner_create_variables /home/XXX/git/jerryscript/jerry-core/parser/js/js-scanner-util.c:2171
    #3 0x4529d4 in parser_push_block_context /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser-statm.c:367
    #4 0x461a95 in parser_parse_for_statement_start /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser-statm.c:1533
    #5 0x461a95 in parser_parse_statements /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser-statm.c:2967
    #6 0x46654b in parser_parse_source /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser.c:2304
    #7 0x466e4b in parser_parse_script /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser.c:2933
    #8 0x4aa2d5 in jerry_parse.constprop.28 /home/XXX/git/jerryscript/jerry-core/api/jerry.c:451
    #9 0x402eb8 in main /home/XXX/git/jerryscript/jerry-main/main-unix.c:112
    #10 0x7fe12c3ca83f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2083f)
    #11 0x4041c8 in _start (/home/XXX/git/jerryscript/build/bin/jerry+0x4041c8)

0x608000000075 is located 0 bytes to the right of 85-byte region [0x608000000020,0x608000000075)
allocated by thread T0 here:
    #0 0x7fe12c852b90 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb90)
    #1 0x4ad294 in jerry_port_read_source /home/XXX/git/jerryscript/jerry-port/default/default-module.c:58
    #2 0x7fe12c3ca83f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2083f)

SUMMARY: AddressSanitizer: heap-buffer-overflow (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xafbf4)
Shadow bytes around the buggy address:
  0x0c107fff7fb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff7fc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff7fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c107fff8000: fa fa fa fa 00 00 00 00 00 00 00 00 00 00[05]fa
  0x0c107fff8010: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c107fff8020: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c107fff8030: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c107fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c107fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==26638==ABORTING
```

Or, if I simplify as this.
```js
a => AFL ;
a . a `a` ;
export default a ;
+ eval `a` ;
for ( let
``` 
I will get this.
```bash
ASAN:DEADLYSIGNAL
=================================================================
==18556==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7feef1c3fb51 bp 0x7ffc3c4431c0 sp 0x7ffc3c442910 T0)
==18556==The signal is caused by a READ memory access.
==18556==Hint: address points to the zero page.
    #0 0x7feef1c3fb50  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xafb50)
    #1 0x4850b1 in lexer_construct_literal_object /home/XXX/git/jerryscript/jerry-core/parser/js/js-lexer.c:2450
    #2 0x485cb4 in scanner_create_variables /home/XXX/git/jerryscript/jerry-core/parser/js/js-scanner-util.c:2171
    #3 0x4529d4 in parser_push_block_context /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser-statm.c:367
    #4 0x461a95 in parser_parse_for_statement_start /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser-statm.c:1533
    #5 0x461a95 in parser_parse_statements /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser-statm.c:2967
    #6 0x46654b in parser_parse_source /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser.c:2304
    #7 0x466e4b in parser_parse_script /home/XXX/git/jerryscript/jerry-core/parser/js/js-parser.c:2933
    #8 0x4aa2d5 in jerry_parse.constprop.28 /home/XXX/git/jerryscript/jerry-core/api/jerry.c:451
    #9 0x402eb8 in main /home/XXX/git/jerryscript/jerry-main/main-unix.c:112
    #10 0x7feef17e683f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2083f)
    #11 0x4041c8 in _start (/home/XXX/git/jerryscript/build/bin/jerry+0x4041c8)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xafb50)
==18556==ABORTING
```

ossy-szeged, 2020-09-29:
Still valid on latest master ( 0ffe1665bd0ee457dd4c1083811e1592a5e2908d ).
The test case hits the same assertion as the test case in #4190 issue:
ICE: Assertion 'context_p->next_scanner_info_p->type == SCANNER_TYPE_BLOCK' failed at /home/oszi/jerryscript/jerry-core/parser/js/js-parser-statm.c(parser_push_block_context):344.

rerobika, 2021-01-18:
Resolved via #4462.

closed 2021-01-18 by rerobika