jerryscript-project/jerryscript #4205: Crash on printing a backtrace from an imported function

amlt, 2020-09-19:
###### JerryScript revision
5555453 (latest master)

###### Build platform
Arch Linux (Linux 5.8.8-zen1-1-zen x86_64)

###### Build steps
```sh
tools/build.py --clean --debug --error-messages=on --line-info=on --logging=on
```

###### Test case
main.c:
```c
#include <stdio.h>
#include <stdlib.h>
#include "jerryscript.h"
#include "jerryscript-port-default.h"
#include "jerryscript-ext/handler.h"

static jerry_value_t handler_trace(const jerry_value_t func_obj,
                                   const jerry_value_t this,
                                   const jerry_value_t argv[],
                                   const jerry_length_t argc)
{
    if (!jerry_is_feature_enabled(JERRY_FEATURE_LINE_INFO))
        printf("Backtrace not available\n");

    jerry_value_t backtrace_array = jerry_get_backtrace(0);
    int array_length = jerry_get_array_length(backtrace_array);

    for (int i = 0; i < array_length; i++) {
        jerry_char_t buf[256];
        jerry_value_t e = jerry_get_property_by_index(backtrace_array, i);
        jerry_size_t copied_bytes = jerry_substring_to_utf8_char_buffer(e,
                0, sizeof(buf) - 1, buf, sizeof(buf) - 1);
        buf[copied_bytes] = '\0';
        printf("%d: %s\n", i, buf);
        jerry_release_value(e);
    }

    jerry_release_value(backtrace_array);

    return jerry_create_undefined();
}

int main()
{
    jerry_char_t resource_name[] = "main.js";
    size_t resource_name_size = sizeof(resource_name) - 1;
    jerry_char_t *script;
    size_t script_size;

    jerry_init(JERRY_INIT_EMPTY);

    jerryx_handler_register_global((const jerry_char_t *)"trace", handler_trace);

    script = jerry_port_read_source(resource_name, &script_size);

    jerry_value_t parsed_code = jerry_parse(resource_name,
                                            resource_name_size,
                                            script,
                                            script_size,
                                            JERRY_PARSE_NO_OPTS);

    if (!jerry_value_is_error(parsed_code)) {
        jerry_value_t ret_value = jerry_run(parsed_code);
        jerry_release_value(ret_value);
    } else {
        printf("parse failed\n");
    }

    jerry_port_release_source(script);

    jerry_release_value(parsed_code);

    jerry_cleanup();
    return 0;
}
```

main.js:
```js
import { my_trace } from "./module.js"
my_trace();
```

module.js:
```js
export function my_trace() {
    trace();
}
```
###### Execution steps
```sh
gcc -g -fsanitize=address -o main main.c -ljerry-core -ljerry-port-default -ljerry-ext -lm
./main
```

###### Output
```
==1214488==ERROR: AddressSanitizer: memcpy-param-overlap: memory ranges [0x558fc8ddbcc9,0x558fc8ddebf1) and [0x558fc8ddbaa4, 0x558fc8dde9cc) overlap
    #0 0x7fa6eedf00ee in __interceptor_memcpy /build/gcc/src/gcc/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:806
    #1 0x558fc8bf665e in ecma_string_copy_to_cesu8_buffer /home/test/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:1068
    #2 0x558fc8bfd74c in ecma_stringbuilder_append /home/test/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:2788
    #3 0x558fc8c9f9e0 in vm_get_backtrace /home/test/jerryscript/jerry-core/vm/vm-utils.c:89
    #4 0x558fc8be44bf in jerry_get_backtrace /home/test/jerryscript/jerry-core/api/jerry.c:4037
    #5 0x558fc8bdc6fe in handler_trace /home/test/jerryscript/tmp/main.c:15
    #6 0x558fc8c33215 in ecma_op_function_call_native /home/test/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1041
    #7 0x558fc8c338b7 in ecma_op_function_call /home/test/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1187
    #8 0x558fc8ca1a75 in opfunc_call /home/test/jerryscript/jerry-core/vm/vm.c:791
    #9 0x558fc8cb7a1f in vm_execute /home/test/jerryscript/jerry-core/vm/vm.c:4789
    #10 0x558fc8cb7fa6 in vm_run /home/test/jerryscript/jerry-core/vm/vm.c:4891
    #11 0x558fc8c32fb9 in ecma_op_function_call_simple /home/test/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:976
    #12 0x558fc8c33896 in ecma_op_function_call /home/test/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1183
    #13 0x558fc8ca1a75 in opfunc_call /home/test/jerryscript/jerry-core/vm/vm.c:791
    #14 0x558fc8cb7a1f in vm_execute /home/test/jerryscript/jerry-core/vm/vm.c:4789
    #15 0x558fc8cb7fa6 in vm_run /home/test/jerryscript/jerry-core/vm/vm.c:4891
    #16 0x558fc8ca0395 in vm_run_global /home/test/jerryscript/jerry-core/vm/vm.c:340
    #17 0x558fc8bdd74c in jerry_run /home/test/jerryscript/jerry-core/api/jerry.c:589
    #18 0x558fc8bdcae8 in main /home/test/jerryscript/tmp/main.c:53
    #19 0x7fa6eeacf151 in __libc_start_main (/usr/lib/libc.so.6+0x28151)
    #20 0x558fc8bdc55d in _start (/home/test/jerryscript/tmp/main+0x4e55d)

0x558fc8ddbcc9 is located 1257 bytes inside of global variable 'jerry_global_heap' defined in '/home/test/jerryscript/jerry-core/jcontext/jcontext.c:139:13' (0x558fc8ddb7e0) of size 524288
0x558fc8ddbaa4 is located 708 bytes inside of global variable 'jerry_global_heap' defined in '/home/test/jerryscript/jerry-core/jcontext/jcontext.c:139:13' (0x558fc8ddb7e0) of size 524288
SUMMARY: AddressSanitizer: memcpy-param-overlap /build/gcc/src/gcc/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:806 in __interceptor_memcpy
==1214488==ABORTING
```

###### Expected behavior
Print a JavaScript backtrace and not crash.

galpeter, 2020-09-21:
Thank you for the report! This is indeed a bug. I have a possible fix for it, will upload it.

galpeter, 2020-09-29:
The #4205 (now merged) PR should fix this.

amlt, 2020-09-29:
I can confirm it is working now. Thank you!

closed 2020-09-29 by amlt