jerryscript-project/jerryscript #4220: Assertion 'jmem_is_heap_pointer (pointer_p)' in jmem_compress_pointer

ossy-szeged, 2020-09-23:
###### JerryScript revision
3c723c9bc1aec08a54bc44bba7987e4324f440e4
(latest master - 2020.09.21.)

###### Build platform
Ubuntu 18.04.5 LTS (Linux 4.15.0-112-generic x86_64)

###### Build steps
tools/build.py --debug --logging=on --error-messages=on --line-info=on

###### Test case
found by test262-esnext tests:
-  Many (48) language/module-code/top-level-await tests
eg:
- language/module-code/top-level-await/syntax/early-no-escaped-await.js
- language/module-code/top-level-await/syntax/export-class-decl-await-expr-array-literal.js
- ...

###### Output
```
ICE: Assertion 'jmem_is_heap_pointer (pointer_p)' failed at /home/oszi/jerryscript/jerry-core/jmem/jmem-allocator.c(jmem_compress_pointer):182.
Error: ERR_FAILED_INTERNAL_ASSERTION
```
###### Expected behavior
not assert

similar issue : #4215

ossy-szeged, 2020-09-30:
still valid on latest master ( 4b2dbd5c2165f60e770e400d56c90d23fa224048 )

ossy-szeged, 2020-12-15:
Not valid now, I'm going to bisect which change fixed it and then close this issue.

rerobika, 2021-01-08:
Closed via https://github.com/jerryscript-project/jerryscript/pull/4266

closed 2021-01-08 by rerobika