jerryscript-project/jerryscript #4237: Assertion context_p->next_scanner_info_p->type == SCANNER_TYPE_FUNCTION in parser_parse_function_arguments

renatahodovan, 2020-09-25:
###### JerryScript revision
629a0e51

###### Build platform
Linux-5.4.0-42-generic-x86_64-with-glibc2.29

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset \
--builddir=build --stack-limit=16
```

###### Test case
```javascript
class C {
    [fxy]() {}
    static String["a"](		
```

###### Output
```text
ICE: Assertion 'context_p->next_scanner_info_p->type == SCANNER_TYPE_FUNCTION' failed at jerryscript/jerry-core/parser/js/js-parser.c(parser_parse_function_arguments):1787.
Error: ERR_FAILED_INTERNAL_ASSERTION
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1624862==ERROR: AddressSanitizer: ABRT on unknown address 0x0018cb1e (pc 0xf7f60b49 bp 0xffd3469c sp 0xffd34680 T0)
    #0 0xf7f60b48  (linux-gate.so.1+0xb48)
    #1 0xf77534c5 in raise (/lib/i386-linux-gnu/libc.so.6+0x354c5)
    #2 0xf773b3f6 in abort (/lib/i386-linux-gnu/libc.so.6+0x1d3f6)
    #3 0x56786077 in jerry_port_fatal jerryscript/jerry-port/default/default-fatal.c:30
    #4 0x566a0c25 in jerry_fatal jerryscript/jerry-core/jrt/jrt-fatals.c:63
    #5 0x566a0c66 in jerry_assert_fail jerryscript/jerry-core/jrt/jrt-fatals.c:87
    #6 0x566ad345 in parser_parse_function_arguments jerryscript/jerry-core/parser/js/js-parser.c:1787
    #7 0x566b29ef in parser_parse_function jerryscript/jerry-core/parser/js/js-parser.c:2579
    #8 0x56751748 in lexer_construct_function_object jerryscript/jerry-core/parser/js/js-lexer.c:2849
    #9 0x56759ce6 in parser_parse_class_literal jerryscript/jerry-core/parser/js/js-parser-expr.c:868
    #10 0x5675a747 in parser_parse_class jerryscript/jerry-core/parser/js/js-parser-expr.c:1002
    #11 0x5677a6ea in parser_parse_statements jerryscript/jerry-core/parser/js/js-parser-statm.c:2911
    #12 0x566b029a in parser_parse_source jerryscript/jerry-core/parser/js/js-parser.c:2317
    #13 0x566b4988 in parser_parse_script jerryscript/jerry-core/parser/js/js-parser.c:3099
    #14 0x56613c57 in jerry_parse jerryscript/jerry-core/api/jerry.c:451
    #15 0x5660d881 in main jerryscript/jerry-main/main-unix.c:112
    #16 0xf773cee4 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x1eee4)
    #17 0x5660d144 in _start (jerryscript/build/bin/jerry+0x1b144)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: ABRT (linux-gate.so.1+0xb48) 
==1624862==ABORTING
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator).</sup>

zherczeg, 2020-10-19:
I get a `SyntaxError: Expected ';' token.`. Maybe it is invalid.

ossy-szeged, 2020-10-19:
Already fixed by bc64957d1999a1db6d669e49f7ec00a6f3b5d9d6.

closed 2020-10-19 by ossy-szeged