jerryscript-project/jerryscript #4240: Date interface returns date without timezone offset on windows platform

godmiaozi, 2020-09-27:
Date interface works fine on ubuntu, but not on windows platform
# on ubuntu
```
jerry> var a = new Date()
undefined
jerry> a
Sun Sep 27 2020 11:08:36 GMT+0800
```

# windows platform
```
jerry> var a = new Date()
undefined
jerry> a
Sun Sep 27 2020 03:10:05 GMT+00:00
jerry>
```

ossy-szeged, 2020-09-27:
Timezone handling should work fine on Windows too a year ago. (And it works for me locally.)
2b3faf683d5f78e696ad97999b761b53f3fed0ae implemented it and was included in 2.1.0 release.

If you still see this issue on the latest JerryScript, please tell us your JerryScript version (release or GIT hash),
details about your OS environment (OS version, local timezone, etc.)

ossy-szeged, 2020-10-07:
@godmiaozi Can you confirm if it works for you too?

godmiaozi, 2020-10-08:
@ossy-szeged 
Thanks for your reply and sorry for reply you so late. I have figured out why it does not work for me. I'm using MINGW to cross compile jerry, which tries to compile the windows compatible executable from linux platform. In this case, MINGW does not define the macro `_WINDOWS`, so the Date interface does not work correctly on my side. But it seems that  `_WIN32` is defined by MINGW. i don't know whether it is better for jerry  to switch from `_WINDOWS` to `_WIN32` or not.

ossy-szeged, 2020-10-12:
@godmiaozi: Thanks for the investigation, I got it. Fix is coming soon.

rerobika, 2021-01-08:
@ossy-szeged What is the progress on this error?

ossy-szeged, 2021-01-08:
> @ossy-szeged What is the progress on this error?

Ah, I forgot it, thanks for the ping. I'll prepare the PR.

PR #4449 (2021-01-13): Fix the build and timezone offset with MinGW
Changes:
- Use _WIN32 macro instead of _WINDOWS
- Fix build error - "no previous declaration for 'UnixTimeToFileTime'"
- Fix conversion build errors
- Remove useless _tzset() call
- Use GetSystemTimeAsFileTime() on Windows to get current time
- Unify and simplify time conversion Windows helper functions

Fixes #4240

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác csaba.osztrogonac@h-lab.eu
commit c84afefd (2021-01-13): Fix the build and timezone offset with MinGW (#4449)

Changes:
- Use _WIN32 macro instead of _WINDOWS
- Fix build error - "no previous declaration for 'UnixTimeToFileTime'"
- Fix conversion build errors
- Remove useless _tzset() call
- Use GetSystemTimeAsFileTime() on Windows to get current time
- Unify and simplify time conversion Windows helper functions

Fixes #4240

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác csaba.osztrogonac@h-lab.eu
closed 2021-01-13 by rerobika