jerryscript-project/jerryscript #4242: Stack-overflow inside function vm_run

HongxuChen, 2020-09-28:
###### JerryScript revision
f58f1507

###### Build platform
Ubuntu 20.04.1 LTS (Linux 5.4.0-48-generic x86_64)

###### Build steps
```sh
mkdir build && cd build && cmake -DCMAKE_BUILD_TYPE=Debug .. && make -j8 
```

###### Test case
```javascript
// test.js
function echo(m) {
    try {
        print(m);
    } catch (e) {
        console.log(m);
    }
}

function oos() {
    oos();
}

try {
    try {
        oos();
    } finally {
        try {
            oos();
        } catch (e) {
            echo("caught: " + e);
        } finally {
        }
    }
} catch (e) {
    if (e) {
        echo("pass");
    }
}
```

###### Execution steps
```sh
./bin/jerry test.js
```

###### Output
Abort without outputs.

###### Backtrace
```
......
#5111 0x00005555555fc813 in ecma_op_function_call_simple (func_obj_p=0x5555557a92d8 <jerry_global_heap+344>, this_arg_value=0x48, arguments_list_p=0x7fffffffb9cc, arguments_list_len=0x0) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-function-object.c:979
#5112 0x00005555555fd1d6 in ecma_op_function_call (func_obj_p=0x5555557a92d8 <jerry_global_heap+344>, this_arg_value=0x48, arguments_list_p=0x7fffffffb9cc, arguments_list_len=0x0) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-function-object.c:1181
#5113 0x000055555566d2c4 in opfunc_call (frame_ctx_p=0x7fffffffb980) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/vm/vm.c:803
#5114 0x0000555555683d44 in vm_execute (frame_ctx_p=0x7fffffffb980) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/vm/vm.c:4882
#5115 0x0000555555684359 in vm_run (shared_p=0x7fffffffbb40, this_binding_value=0x1b, lex_env_p=0x5555557a91b0 <jerry_global_heap+48>) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/vm/vm.c:4983
#5116 0x00005555555fc813 in ecma_op_function_call_simple (func_obj_p=0x5555557a92d8 <jerry_global_heap+344>, this_arg_value=0x48, arguments_list_p=0x7fffffffbd38, arguments_list_len=0x0) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-function-object.c:979
#5117 0x00005555555fd1d6 in ecma_op_function_call (func_obj_p=0x5555557a92d8 <jerry_global_heap+344>, this_arg_value=0x48, arguments_list_p=0x7fffffffbd38, arguments_list_len=0x0) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-function-object.c:1181
#5118 0x000055555566d2c4 in opfunc_call (frame_ctx_p=0x7fffffffbce0) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/vm/vm.c:803
#5119 0x0000555555683d44 in vm_execute (frame_ctx_p=0x7fffffffbce0) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/vm/vm.c:4882
#5120 0x0000555555684359 in vm_run (shared_p=0x7fffffffbe70, this_binding_value=0x1b, lex_env_p=0x5555557a91b0 <jerry_global_heap+48>) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/vm/vm.c:4983
#5121 0x000055555566b977 in vm_run_global (bytecode_p=0x5555557a9590 <jerry_global_heap+1040>) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/vm/vm.c:339
#5122 0x00005555555a380b in jerry_run (func_val=0x12b) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/api/jerry.c:593
#5123 0x000055555559dbf6 in main (argc=0x2, argv=0x7fffffffc4d8) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-main/main-unix.c:123
```

###### Expected behavior
Jerryscript does not check the limit of the call stack. I think It should report some error messages like `jsc` or `node`:
```
caught: RangeError: Maximum call stack size exceeded.
pass
```

rerobika, 2020-09-28:
Hi @HongxuChen!

The stack limit check is not enabled by the engine default but you can enable it by passing `-DJERRY_STACK_LIMIT=(KB)` as a CMake option. In this case the expected RangeError is thrown.

closed 2020-09-28 by rerobika

HongxuChen, 2020-09-28:
@rerobika Thanks for pointing this!
I noticed that when stack limit is set to 512kb, the first example in #4243 still causes stack overflow; while for 256kb, it won't. Additionally, when stack limit is 1000kb, it will crash the test file in this issue. May I know what's the suggested maximal stack when using jerryscript?