jerryscript-project/jerryscript #4243: Stack-overflow at ecma-proxy-object.c:277 (ecma_proxy_object_get)

HongxuChen, 2020-09-28:
###### JerryScript revision
f58f1507

###### Build platform
Ubuntu 20.04.1 LTS (Linux 5.4.0-48-generic x86_64)

###### Build steps
```sh
mkdir build && cd build && cmake -DCMAKE_BUILD_TYPE=Debug .. && make -j8 
```

###### Test case
```javascript
// test.js
h = {}
p = new Proxy({}, h)
h.__proto__ = p;
try { p._ } catch {}
```

###### Execution steps
```sh
./bin/jerry test.js
```

###### Output
Abort (with regular build) without outputs.

###### Backtrace
```
#0  __GI_raise (sig=sig@entry=0x6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff73a5859 in __GI_abort () at abort.c:79
#2  0x00005555556fe304 in jerry_port_fatal (code=ERR_REF_COUNT_LIMIT) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-port/default/default-fatal.c:30
#3  0x000055555562d2a0 in jerry_fatal (code=ERR_REF_COUNT_LIMIT) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/jrt/jrt-fatals.c:63
#4  0x00005555555ac8e2 in ecma_ref_object (object_p=0x5555557a92e8 <jerry_global_heap+360>) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/base/ecma-gc.c:138
#5  0x00005555555c531e in ecma_copy_value (value=0x16b) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/base/ecma-helpers-value.c:913
#6  0x00005555555f6c2e in ecma_op_to_object (value=0x16b) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-conversion.c:545
#7  0x000055555560aae8 in ecma_op_get_method (value=0x16b, prop_name_p=0x2ad) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:894
#8  0x000055555560ac1c in ecma_op_get_method_by_magic_id (value=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:967
#9  0x0000555555614790 in ecma_validate_proxy_object (handler=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:277
#10 0x0000555555616a41 in ecma_proxy_object_get (obj_p=0x5555557a9318 <jerry_global_heap+408>, prop_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:1128
#11 0x000055555560a688 in ecma_op_object_get_with_receiver (object_p=0x5555557a9318 <jerry_global_heap+408>, property_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:751
#12 0x000055555560ab41 in ecma_op_object_get (property_name_p=0x2ad, object_p=0x5555557a92e8 <jerry_global_heap+360>)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:724
#13 ecma_op_get_method (value=0x16b, prop_name_p=0x2ad) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:904
#14 0x000055555560ac1c in ecma_op_get_method_by_magic_id (value=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:967
#15 0x0000555555614790 in ecma_validate_proxy_object (handler=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:277
#16 0x0000555555616a41 in ecma_proxy_object_get (obj_p=0x5555557a9318 <jerry_global_heap+408>, prop_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:1128
#17 0x000055555560a688 in ecma_op_object_get_with_receiver (object_p=0x5555557a9318 <jerry_global_heap+408>, property_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:751
#18 0x000055555560ab41 in ecma_op_object_get (property_name_p=0x2ad, object_p=0x5555557a92e8 <jerry_global_heap+360>)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:724
#19 ecma_op_get_method (value=0x16b, prop_name_p=0x2ad) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:904
#20 0x000055555560ac1c in ecma_op_get_method_by_magic_id (value=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:967
#21 0x0000555555614790 in ecma_validate_proxy_object (handler=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:277
#22 0x0000555555616a41 in ecma_proxy_object_get (obj_p=0x5555557a9318 <jerry_global_heap+408>, prop_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:1128
#23 0x000055555560a688 in ecma_op_object_get_with_receiver (object_p=0x5555557a9318 <jerry_global_heap+408>, property_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:751
#24 0x000055555560ab41 in ecma_op_object_get (property_name_p=0x2ad, object_p=0x5555557a92e8 <jerry_global_heap+360>)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:724
#25 ecma_op_get_method (value=0x16b, prop_name_p=0x2ad) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:904
#26 0x000055555560ac1c in ecma_op_get_method_by_magic_id (value=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:967
#27 0x0000555555614790 in ecma_validate_proxy_object (handler=0x16b, magic_id=LIT_MAGIC_STRING_GET)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:277
#28 0x0000555555616a41 in ecma_proxy_object_get (obj_p=0x5555557a9318 <jerry_global_heap+408>, prop_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-proxy-object.c:1128
#29 0x000055555560a688 in ecma_op_object_get_with_receiver (object_p=0x5555557a9318 <jerry_global_heap+408>, property_name_p=0x2ad, receiver=0x16b)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:751
#30 0x000055555560ab41 in ecma_op_object_get (property_name_p=0x2ad, object_p=0x5555557a92e8 <jerry_global_heap+360>)
    at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:724
#31 ecma_op_get_method (value=0x16b, prop_name_p=0x2ad) at /home/hongxu/work/jerryscript/jerryscript-asan/jerry-core/ecma/operations/ecma-objects.c:904
```

###### Expected behavior
No errors.

HongxuChen, 2020-09-28:
Similar proxy objects can also be found.

```javascript
p = new Proxy({}, {});
p.__proto__ = p
Reflect.has(p)
```
This causes stack-overflow at `ecma-proxy-object.c:1024 (ecma_proxy_object_has)`
```
#0  0x00005555555707d5 in ecma_is_lexical_environment (object_p=<error reading variable: Cannot access memory at address 0x7fffff7feff8>) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/base/ecma-helpers.c:174
#1  0x0000555555570968 in ecma_get_object_is_builtin (object_p=0x5555556472b0 <jerry_global_heap+8>) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/base/ecma-helpers.c:218
#2  0x000055555557a21c in ecma_builtin_try_to_instantiate_property (object_p=0x5555556472b0 <jerry_global_heap+8>, string_p=0x2cd) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/builtin-objects/ecma-builtins.c:780
#3  0x0000555555591b42 in ecma_op_object_find_own (base_value=0x153, object_p=0x5555556472b0 <jerry_global_heap+8>, property_name_p=0x2cd) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:572
#4  0x0000555555591e53 in ecma_op_object_get_with_receiver (object_p=0x5555556472b0 <jerry_global_heap+8>, property_name_p=0x2cd, receiver=0x153) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:755
#5  0x0000555555592282 in ecma_op_object_get (property_name_p=0x2cd, object_p=0x5555556473f8 <jerry_global_heap+336>) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:724
#6  ecma_op_get_method (value=0x153, prop_name_p=0x2cd) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:904
#7  0x000055555559235d in ecma_op_get_method_by_magic_id (value=0x153, magic_id=LIT_MAGIC_STRING_HAS) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:967
#8  0x00005555555978b9 in ecma_validate_proxy_object (handler=0x153, magic_id=LIT_MAGIC_STRING_HAS) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-proxy-object.c:277
#9  0x0000555555598822 in ecma_proxy_object_has (obj_p=0x555555647400 <jerry_global_heap+344>, prop_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-proxy-object.c:1024
#10 0x0000555555591434 in ecma_op_object_has_property (object_p=0x555555647400 <jerry_global_heap+344>, property_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:372
#11 0x0000555555598875 in ecma_proxy_object_has (obj_p=0x555555647400 <jerry_global_heap+344>, prop_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-proxy-object.c:1038
#12 0x0000555555591434 in ecma_op_object_has_property (object_p=0x555555647400 <jerry_global_heap+344>, property_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:372
#13 0x0000555555598875 in ecma_proxy_object_has (obj_p=0x555555647400 <jerry_global_heap+344>, prop_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-proxy-object.c:1038
#14 0x0000555555591434 in ecma_op_object_has_property (object_p=0x555555647400 <jerry_global_heap+344>, property_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:372
#15 0x0000555555598875 in ecma_proxy_object_has (obj_p=0x555555647400 <jerry_global_heap+344>, prop_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-proxy-object.c:1038
#16 0x0000555555591434 in ecma_op_object_has_property (object_p=0x555555647400 <jerry_global_heap+344>, property_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:372
#17 0x0000555555598875 in ecma_proxy_object_has (obj_p=0x555555647400 <jerry_global_heap+344>, prop_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-proxy-object.c:1038
#18 0x0000555555591434 in ecma_op_object_has_property (object_p=0x555555647400 <jerry_global_heap+344>, property_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:372
#19 0x0000555555598875 in ecma_proxy_object_has (obj_p=0x555555647400 <jerry_global_heap+344>, prop_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-proxy-object.c:1038
#20 0x0000555555591434 in ecma_op_object_has_property (object_p=0x555555647400 <jerry_global_heap+344>, property_name_p=0x1f2d) at /home/hongxu/work/jerryscript/jerryscript-orig/jerry-core/ecma/operations/ecma-objects.c:372
```

rerobika, 2020-09-28:
Duplicate of #4242 since these are simple stack overflow issues.

closed 2020-09-28 by rerobika