jerryscript-project/jerryscript #4244: Out-of-bound read in main_print_unhandled_exception

paintedveil5, 2020-09-29:
##### JerryScript revision
0ffe1665bd0ee457dd4c1083811e1592a5e2908d
(latest master - 2020.09.28.)

###### Build platform
Ubuntu 16.04.1 (Linux 4.15.0-117-generic x86_64)

###### Build steps

```sh
python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --lto=off --error-message=on --system-allocator=on --compile-flag=-Wno-maybe-uninitializ
```
###### Test case
` //debugger.js
 try {
 } `

###### Execution steps

```sh
./jerry debugger.js
```
###### Output
`
READ of size 1 at 0xf640077b thread T0
    #0 0x804ac66 in main_print_unhandled_exception /home/lily/Desktop/debugger/jerryscript/jerry-main/main-utils.c:196
    #1 0x80498fc in main /home/lily/Desktop/debugger/jerryscript/jerry-main/main-unix.c:142
    #2 0xf782b646 in __libc_start_main (/lib32/libc.so.6+0x18646)
    #3 0x8049320  (/home/lily/Desktop/debugger/jerryscript/build/bin/jerry+0x8049320)

0xf640077b is located 0 bytes to the right of 27-byte region [0xf6400760,0xf640077b)
allocated by thread T0 here:
    #0 0xf7a5ddee in malloc (/usr/lib32/libasan.so.2+0x96dee)
    #1 0x819498a in jerry_port_read_source /home/lily/Desktop/debugger/jerryscript/jerry-port/default/default-module.c:58
    #2 0x804aaa7 in main_print_unhandled_exception /home/lily/Desktop/debugger/jerryscript/jerry-main/main-utils.c:164
    #3 0x80498fc in main /home/lily/Desktop/debugger/jerryscript/jerry-main/main-unix.c:142
    #4 0xf782b646 in __libc_start_main (/lib32/libc.so.6+0x18646)
`

###### Expected behavior
Please consider that the error line may be larger than the test case line number.  The above "try" test case has just three lines, but the error happened on line 4 (a "catch" block is needed. ) The **jerry_port_read_source (path_str_p, &source_size)** called in the **main_print_unhandled_exception** need to malloc more memory to locate an error happened out of the source_size.

zherczeg, 2020-09-29:
Thank you for analyzing the issue, few people are doing this. This is a great help for us!

ossy-szeged, 2020-09-29:
It is the same bug as #4214, but I leave this one open.
Additionaly info from the other issue:

affected test262-esnext tests:
    language/asi/S7.9_A5.7_T1.js
    language/asi/S7.9_A9_T6.js
    language/asi/S7.9_A9_T7.js
    language/statements/try/S12.14_A16_T1.js

@dbatyai : This part of the source was refactored by edab196.

closed 2020-12-01 by rerobika

zherczeg, 2020-12-19:
https://nvd.nist.gov/vuln/detail/CVE-2020-29657
9.1 critical
seriously...

rzr, 2021-01-03:
Also reported at:

https://security-tracker.debian.org/tracker/CVE-2020-29657

https://github.com/jerryscript-project/iotjs/issues/1955

PR #4350 (2020-12-01): Correct syntax error printing in case of end of file
The old variant of the syntax error printing method could incorrectly over-read the source file contents.

Fixes: #4244