jerryscript-project/jerryscript #4263: jerry-script snapshot documentation doesn't explain how to generate snapshot for function (-f)

jvermillard, 2020-10-01:
I spend some time trying to understand how to use `jerry-generate` to generate bycode for functions and not scripts 
(jerry_generate_function_snapshot vs jerry_generate_snapshot)

the jerry-generate --help is not really helpful:
```
./jerry-snapshot --help                                                       jerryscript/git/master !
./jerry-snapshot [-h] [COMMAND]

  -h, --help            print this help and exit
  COMMAND               specify the command

Available commands:
  generate
  litdump
  merge

Passing -h or --help after a command displays its help.
```

To figure I need to use: 
`./jerry-snapshot generate ./function.js -f _function`

I think the `-f` option is explained nowhere in the doc, I figured it from the code.

###### JerryScript revision
master: 0ffe1665

rerobika, 2020-10-01:
Hi @jvermillard!

Passing `--help` after `generate` cli options gives you more detailed information about the capabilities of the snapshot tool generator part.

```
$ build/bin/jerry-snapshot generate --help
build/bin/jerry-snapshot generate [-h] [-s] [-f ARGUMENTS] [--load-literals-list-format FILE]
  [--show-opcodes] [-o FILE] [FILE]

  -h, --help            print this help and exit
  -s, --static          generate static snapshot
  -f, --generate-function-snapshot ARGUMENTS
                        generate function snapshot with given arguments
  --load-literals-list-format FILE
                        import literals from list format (for static snapshots)
  --show-opcodes        print generated opcodes
  -o FILE               specify output file name (default: js.snapshot)
  FILE                  input source file

```

jvermillard, 2020-10-01:
damn I tried `jerry-snapshot --help generate` and `jerry-snapshot help generate` not this one and wait straight ot the code :facepalm:  

thanks

closed 2020-10-01 by jvermillard