jerryscript-project/jerryscript #4295: Problem Compile Under Ubuntu x86-64 [Clang 10.0]

Andres6936, 2020-10-18:
```shellsession
andres6936@Tierra ~/CLionProjects/jerryscript/cmake-build-debug $ uname -a
Linux Tierra 4.15.0-46-generic #49~16.04.1-Ubuntu SMP Tue Feb 12 17:45:24 UTC 2019 x86_64 GNU/Linux
```

```shellsession
andres6936@Tierra ~/CLionProjects/jerryscript/cmake-build-debug $ clang -v
clang version 10.0.0 
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/local/bin
```

```shellsession
andres6936@Tierra ~/CLionProjects/jerryscript/cmake-build-debug $ cmake ..
-- The C compiler identification is Clang 10.0.0
-- Check for working C compiler: /usr/local/bin/clang
-- Check for working C compiler: /usr/local/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- CMAKE_BUILD_TYPE            MinSizeRel
-- CMAKE_C_COMPILER_ID         Clang
-- CMAKE_SYSTEM_NAME           Linux
-- CMAKE_SYSTEM_PROCESSOR      x86_64
-- BUILD_SHARED_LIBS           OFF
-- ENABLE_LTO                  ON
-- ENABLE_STRIP                ON
-- JERRY_CMDLINE               ON
-- JERRY_CMDLINE_TEST          OFF
-- JERRY_CMDLINE_SNAPSHOT      OFF
-- JERRY_LIBFUZZER             OFF
-- JERRY_PORT_DEFAULT          ON (FORCED BY CMDLINE OR LIBFUZZER OR TESTS)
-- JERRY_EXT                   ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_LIBM                  ON
-- UNITTESTS                   OFF
-- DOCTESTS                    OFF
-- ENABLE_ALL_IN_ONE              OFF
-- JERRY_CPOINTER_32_BIT          OFF
-- JERRY_DEBUGGER                 OFF
-- JERRY_ERROR_MESSAGES           OFF
-- JERRY_EXTERNAL_CONTEXT         OFF
-- JERRY_PARSER                   ON
-- JERRY_LINE_INFO                OFF
-- JERRY_LOGGING                  OFF
-- JERRY_MEM_STATS                OFF
-- JERRY_MEM_GC_BEFORE_EACH_ALLOC OFF
-- JERRY_PARSER_DUMP_BYTE_CODE    OFF
-- JERRY_PROFILE                  es.next
-- JERRY_REGEXP_STRICT_MODE       OFF
-- JERRY_REGEXP_DUMP_BYTE_CODE    OFF
-- JERRY_SNAPSHOT_EXEC            OFF
-- JERRY_SNAPSHOT_SAVE            OFF
-- JERRY_SYSTEM_ALLOCATOR         OFF
-- JERRY_VALGRIND                 OFF
-- JERRY_VM_EXEC_STOP             OFF
-- JERRY_GLOBAL_HEAP_SIZE         (512)
-- JERRY_GC_LIMIT                 (0)
-- JERRY_STACK_LIMIT              (0)
-- JERRY_GC_MARK_LIMIT            (8)
-- FEATURE_INIT_FINI           OFF
-- Performing Test HAVE_TM_GMTOFF
-- Performing Test HAVE_TM_GMTOFF - Failed
-- Looking for include file time.h
-- Looking for include file time.h - not found
-- Looking for include file unistd.h
-- Looking for include file unistd.h - not found
-- ENABLE_LINK_MAP             OFF
-- JERRY_TEST_STACK_MEASURE    OFF
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andres6936/CLionProjects/jerryscript/cmake-build-debug
```

```shellsession
andres6936@Tierra ~/CLionProjects/jerryscript/cmake-build-debug $ make -j2
...
[ 12%] Linking C static library ../lib/libjerry-libm.a
[ 12%] Built target jerry-libm
Scanning dependencies of target jerry-core
...
[ 18%] Building C object jerry-core/CMakeFiles/jerry-core.dir/ecma/base/ecma-helpers-value.c.o
/home/andres6936/CLionProjects/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:564:17: error: implicit conversion from 'ecma_length_t' (aka 'unsigned long') to 'double' may lose precision [-Werror,-Wimplicit-int-float-conversion]
  JERRY_ASSERT (number <= ECMA_NUMBER_MAX_SAFE_INTEGER);
                ^~~~~~ ~~
/home/andres6936/CLionProjects/jerryscript/jerry-core/jrt/jrt.h:88:21: note: expanded from macro 'JERRY_ASSERT'
      JERRY_UNUSED (x); \
                    ^
/home/andres6936/CLionProjects/jerryscript/jerry-core/jrt/jrt.h:34:34: note: expanded from macro 'JERRY_UNUSED'
#define JERRY_UNUSED(x) ((void) (x))
                                 ^
/home/andres6936/CLionProjects/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:2971:17: error: implicit conversion from 'ecma_length_t' (aka 'unsigned long') to 'double' may lose precision [-Werror,-Wimplicit-int-float-conversion]
  JERRY_ASSERT (index <= ECMA_NUMBER_MAX_SAFE_INTEGER);
                ^~~~~ ~~
/home/andres6936/CLionProjects/jerryscript/jerry-core/jrt/jrt.h:88:21: note: expanded from macro 'JERRY_ASSERT'
      JERRY_UNUSED (x); \
                    ^
/home/andres6936/CLionProjects/jerryscript/jerry-core/jrt/jrt.h:34:34: note: expanded from macro 'JERRY_UNUSED'
#define JERRY_UNUSED(x) ((void) (x))
                                 ^
2 errors generated.
make[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/build.make:219: jerry-core/CMakeFiles/jerry-core.dir/ecma/base/ecma-helpers-string.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/andres6936/CLionProjects/jerryscript/cmake-build-debug'
make[1]: *** [CMakeFiles/Makefile2:235: jerry-core/CMakeFiles/jerry-core.dir/all] Error 2
make[1]: Leaving directory '/home/andres6936/CLionProjects/jerryscript/cmake-build-debug'
make: *** [Makefile:130: all] Error 2
```

**The warnings are being treated as errors, hence the compilation process is over.**

rerobika, 2020-10-19:
Hi @Andres6936!

Please share your git revision since very similar errors have been resolved lately in #4288 .

ossy-szeged, 2020-10-19:
cc-ing @kaadam

ossy-szeged, 2020-10-19:
Already fixed by #4288.

closed 2020-10-19 by ossy-szeged