jerryscript-project/jerryscript #4300: Stack overflow at ecma_op_object_get_with_receiver  jerry-core/ecma/operations/ecma-objects.c:795

kvenux, 2020-10-20:
# Build environment:

Ubuntu 16.04
gcc 5.4.0
jerry version: 6f25d6e
build command: python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --lto=off --error-message=on --system-allocator=on
test command: ./xst poc

# Target device:
Desktop Linux

# POC
[jerry-001361.txt](https://github.com/jerryscript-project/jerryscript/files/5406788/jerry-001361.txt)

# Description
Below is the ASAN outputs.

ASAN:SIGSEGV
=================================================================
==3664==ERROR: AddressSanitizer: stack-overflow on address 0xff502fd0 (pc 0x081262ca bp 0xff503058 sp 0xff502fc0 T0)
    #0 0x81262c9 in ecma_op_object_get_with_receiver /home/keven/Fuzzing/jerry-test/jerry-core/ecma/operations/ecma-objects.c:795
    #1 0x812625c in ecma_op_object_get /home/keven/Fuzzing/jerry-test/jerry-core/ecma/operations/ecma-objects.c:774
    #2 0x80c6f9d in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:625
    #3 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #4 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #5 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #6 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #7 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #8 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #9 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #10 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #11 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #12 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #13 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #14 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #15 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #16 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #17 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #18 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #19 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #20 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #21 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #22 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #23 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #24 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #25 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #26 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #27 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #28 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #29 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #30 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #31 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #32 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #33 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #34 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #35 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #36 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #37 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #38 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #39 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #40 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #41 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #42 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #43 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #44 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #45 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #46 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #47 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #48 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #49 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
[... 179 lines cut ...]
    #229 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
[... cut ...]
    #248 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670
    #249 0x80c6d31 in ecma_builtin_json_internalize_process_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:749
    #250 0x80c7438 in ecma_builtin_json_internalize_property /home/keven/Fuzzing/jerry-test/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:670

SUMMARY: AddressSanitizer: stack-overflow /home/keven/Fuzzing/jerry-test/jerry-core/ecma/operations/ecma-objects.c:795 ecma_op_object_get_with_receiver
==3664==ABORTING

akosthekiss, 2020-10-20:
Thanks for reporting this issue. (And the follow-ups.) But. Before reporting any more stack overflow issues, please, check whether they are also reproducible if JerryScript is built with stack limit. If the SO does not happen with stack limit, the issue is not valid.

kvenux, 2020-10-21:
> Thanks for reporting this issue. (And the follow-ups.) But. Before reporting any more stack overflow issues, please, check whether they are also reproducible if JerryScript is built with stack limit. If the SO does not happen with stack limit, the issue is not valid.

Noted. I will double check

rerobika, 2020-10-22:
Closing the issue since the stack-overflow can be prevented by passing `--stack-limit X` build option. I've tested the issue and the expected RangeError is thrown.

closed 2020-10-22 by rerobika