jerryscript-project/jerryscript #4322: Show opcodes from Snapshot

dakhnod, 2020-11-01:
Is there any way to show debug info (function parameters, op calls etc..) while executing a snapshot? If not, could that be added?

zherczeg, 2020-11-01:
This is not supported yet, but I think it should not be too hard to implement it. Is it a feature request or do you want to work on it?

dakhnod, 2020-11-01:
I will take a look at it, thanks for the quick response.
What IDE did you use for developing?

dakhnod, 2020-11-01:
Regarding that, how can I get a parser_list_t pointer used by parse_print_final_cbc from a ecma_compiled_code_t* thingy?
I seems to me that the literal pool gets build during parsing. I can't find a way to build it from already compiled code...

zherczeg, 2020-11-01:
Oh, you are right.
https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/parser/js/js-parser.c#L607

You need a literal pool to print the names of certain variables. E.g. in case of `function f() { var v1 }`, we don't store `v1` at all because we want to save memory. Whithout that, we probably need another dump function which uses the literal pool to get the names of variables when it is possible. Or extend this dump function to work in two modes: when literal_pool_p is not NULL, and when it is NULL.

dakhnod, 2020-11-01:
Could you please take a look at that? I'd really love to be able to reverse engineer bigger and more complicated snapshots. I did try to figure the jerry code out, but really lack the knowledge...

dakhnod, 2020-11-02:
Is the byte code format documented somewhere? There are some bytes I wasn't able to explain to myself...

zherczeg, 2020-11-04:
Please check #4327

dakhnod, 2020-11-04:
@zherczeg is there any way to convert 2.1.0 code to the latest version without recompiling?
Or would it suffice to just change the version header in that case?

Also, how can I include multiple primary functions?

zherczeg, 2020-11-04:
When the byte code changes (including layout changes), it often just follows a new internal logic / approach, so there is no trivial 1:1 conversion from the old byte code to the new one. Hence we don't develop tools for conversion.

Snapshot tool can merge multiple snapshots into one if that is what you refer to.

dakhnod, 2020-11-10:
Hi Zoltan, could you please pm me? I see no way of contacting you privately...
dakhnod@gmail.com

closed 2020-11-16 by dbatyai