jerryscript-project/jerryscript #4334: Number operations with big numbers are inaccurate on 32 bit

ossy-szeged, 2020-11-12:
###### JerryScript revision
25bb5083be896bb70d175ff9ef2fdd5e8a31ff45
(latest master - 2020.11.10.)

###### Build platform
Ubuntu 18.04.5 LTS (Linux 4.15.0-112-generic x86_64)

###### Build steps
tools/build.py --compile-flag=-m32

###### Test case
test262/esnext
-- language/types/number/S8.5_A2.1.js
-- language/types/number/S8.5_A2.2.js

minimal test case: 9007199254740994 + 0.9999847412109375
(note: 9007199254740994 == 2^53 + 2 and 0.9999847412109375 == 1.0 - 1/65536.0)

###### Output
9007199254740996, but the expected output is 9007199254740994

galpeter, 2020-11-12:
It seems the issue comes from the incorrect double calculation on the 32bit intel systems. So we should run the 32 bit tests with the extra `--compile-flag=-mfpmath=sse --compile-flag=-msse2` options or use the `toolchain_linux_i686.cmake` toolchain file.
( https://github.com/jerryscript-project/jerryscript/blob/master/docs/01.CONFIGURATION.md#x86-with-gcc )

ossy-szeged, 2020-11-12:
See, it's not a bug, but "--compile-flag=-m32" in itself is a non spec conform build option.
It would be great if build.py can automatically add these necessary extra options in this case.

closed 2020-11-12 by ossy-szeged