jerryscript-project/jerryscript #4342: Incorrect result for Bigint left shift

rerobika, 2020-11-17:
###### JerryScript revision
5d916fb8d4cd944ff985bcbc89cfde017425744d 

###### Build platform
Ubuntu 18.04.4 LTS (Linux 5.4.0-42-generic x86_64)

###### Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --stack-limit=16
```

###### Test case
```javascript
-1n << ~(2n ** 32n)
```

###### Output
```javascript
0
```

###### Expected output (verified by V8 and SpiderMonkey)
```javascript
-1n
```

###### Advice for the error investigation:
Potential integer overflow based on these evaluation results. 
```javascript
jerry> -1n << ~(2n ** 32n)
0
jerry> -1n << ~((2n ** 32n) - 1n)
0
jerry> -1n << ~((2n ** 32n) - 2n)
-1
jerry> -1n << ~((2n ** 32n) - 3n)
-1
// -1n << ~(2n ** 32n) -{4, 5, 6 ... } give the correct result
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator).</sup>

rerobika, 2020-11-17:
Note: The existing issue is potentially similar to #4341 .

zherczeg, 2020-11-17:
Obviously since shifting with negative number changes the direction of shift, so both issues refer to right shift.

rerobika, 2021-01-05:
Closing, duplicate of #4341 .

closed 2021-01-05 by rerobika