jerryscript-project/jerryscript #4366: Assertion 'ecma_get_object_type (func_obj_p) == ECMA_OBJECT_TYPE_NATIVE_FUNCTION' in ecma_op_function_get_function_realm

ossy-szeged, 2020-12-15:
###### JerryScript revision
608325903030766f2894ba5641644cb7ccc5d3a4
(latest master - 2020.12.15.)

###### Build platform
Ubuntu 18.04.5 LTS (Linux 4.15.0-112-generic x86_64)

###### Build steps
tools/build.py --debug --logging=on --error-messages=on --line-info=on

###### Test cases
test262-esnext test:
- built-ins/Function/prototype/bind/get-fn-realm-recursive.js
- built-ins/Function/prototype/bind/get-fn-realm.js
- built-ins/Proxy/get-fn-realm.js
- built-ins/TypedArrayConstructors/ctors/typedarray-arg/detached-when-species-retrieved-different-type.js
- built-ins/TypedArrayConstructors/ctors/typedarray-arg/detached-when-species-retrieved-same-type.js
- built-ins/TypedArrayConstructors/ctors-bigint/typedarray-arg/detached-when-species-retrieved-different-type.js
- built-ins/TypedArrayConstructors/ctors-bigint/typedarray-arg/detached-when-species-retrieved-same-type.js

###### Output
```
 ICE: Assertion 'ecma_get_object_type (func_obj_p) == ECMA_OBJECT_TYPE_NATIVE_FUNCTION' failed at /home/oszi/jerryscript/jerry-core/ecma/operations/ecma-function-object.c(ecma_op_function_get_function_realm):806.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

###### Expected behavior
not assert

ossy-szeged, 2020-12-15:
@zherczeg ^^^

rerobika, 2021-01-05:
Related PR #4407.

rerobika, 2021-01-08:
@kisbg Could you please check these tests as well?

```
built-ins/Function/prototype/bind/get-fn-realm-recursive.js
built-ins/Function/prototype/bind/get-fn-realm.js
```

You should add bound function support for ecma_op_function_get_realm.

rerobika, 2021-01-11:
The remaining errors after #4407.
```
built-ins/TypedArrayConstructors/ctors/typedarray-arg/detached-when-species-retrieved-different-type.js
built-ins/TypedArrayConstructors/ctors/typedarray-arg/detached-when-species-retrieved-same-type.js
built-ins/TypedArrayConstructors/ctors-bigint/typedarray-arg/detached-when-species-retrieved-different-type.js
built-ins/TypedArrayConstructors/ctors-bigint/typedarray-arg/detached-when-species-retrieved-same-type.js
```

rerobika, 2021-01-18:
Closing the issue, the remaining issues are irrelevant from the original issue and does not cause assertion error, simply these features are not supprted yet.

closed 2021-01-18 by rerobika