jerryscript-project/jerryscript #4384: test262 are not match unix testing result on windows

lygstate, 2020-12-28:
###### JerryScript revision
9676500addbcc5974bf81f7b3726230d26f5c1a5

###### Build platform

```
C:\Users\lygstate>python -c "import platform; print(platform.platform())"
Windows-10-10.0.19041-SP0
```
###### Build steps
For ES 5.1
```
PS C:\work\study\languages\typescript\jerryscript> python tools/run-tests.py --test262
Build command:
cmd \
        /S \
        /C \
        C:\work\study\languages\typescript\jerryscript\tools\build.py \
        --profile=es5.1 \
        --builddir=C:\work\study\languages\typescript\jerryscript\build\tests\test262_tests \
        --install=C:\work\study\languages\typescript\jerryscript\build\tests\test262_tests\local
Test command:
TZ='America/Los_Angeles' \
cmd \
        /S \
        /C \
        C:\work\study\languages\typescript\jerryscript\tools\runners/run-test-suite-test262.py \
        --engine \
        C:\work\study\languages\typescript\jerryscript\build\tests\test262_tests\local\bin\jerry.exe --test262-object \
        --test-dir \
        C:\work\study\languages\typescript\jerryscript\tests/test262 \
        --es51
.................................................. Executed 5000 tests.
.................................................. Executed 10000 tests.
.................
=== Summary ===
 - Ran 11725 tests
 - Passed 11557 tests (98.6%)
 - Failed 168 tests (1.4%)
```

For ES6
```
PS C:\work\study\languages\typescript\jerryscript> python tools/run-tests.py --test262-es2015 update
Build command:
cmd \
        /S \
        /C \
        C:\work\study\languages\typescript\jerryscript\tools\build.py \
        --profile=es.next \
        --line-info=on \
        --error-messages=on \
        --builddir=C:\work\study\languages\typescript\jerryscript\build\tests\test262_tests_es2015 \   
        --install=C:\work\study\languages\typescript\jerryscript\build\tests\test262_tests_es2015\local
Test command:
TZ='America/Los_Angeles' \
cmd \
        /S \
        /C \
        C:\work\study\languages\typescript\jerryscript\tools\runners/run-test-suite-test262.py \
        --engine \
        C:\work\study\languages\typescript\jerryscript\build\tests\test262_tests_es2015\local\bin\jerry.exe --test262-object \
        --test-dir \
        C:\work\study\languages\typescript\jerryscript\tests/test262 \
        --es2015 \
        update
.................................................. Executed 5000 tests.
.................................................. Executed 10000 tests.
.................................................. Executed 15000 tests.
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-out-kkxx3goy'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-out-x6oy5j1s'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-err-nj697qhg'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-err-ymdteaqd'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-out-5b09cyb1'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-out-10s2ozab'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-err-dzy30erw'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-err-wfrmjsp4'
.................................................. Executed 20000 tests.
.................................................. Executed 25000 tests.
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-out-qu8dgc37'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-out-e_0fjy1l'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-err-k3yhbr7v'
ERROR:root:Error disposing temp file: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'C:\\Users\\lygstate\\AppData\\Local\\Temp\\test262-err-41uefzcl'
.....................................
=== Summary ===
 - Ran 28709 tests
 - Passed 28070 tests (97.8%)
 - Failed 639 tests (2.2%)

=== Summary - updating excludelist ===

New failing tests added to the excludelist
  built-ins/String/raw/special-characters.js
  language/expressions/template-literal/tv-line-continuation.js
  language/expressions/template-literal/tv-line-terminator-sequence.js

Excludelist was updated succesfully.
```

ossy-szeged, 2021-01-08:
I haven't seen any extra failure on Windows with ES5.1, ES2015 and ES.Next version of test262.
Tested on the latest master ( 7b00db40799630709e6017ba9073815a4b6cb4eb ) in cmd.exe not in PS.

There is only one es2015 test hang (infinite loop) in strict and non-strict mode, but it isn't related
to Windows, but a spec change - language/statements/for-of/iterator-next-reference.js.

lygstate, 2021-01-09:
> I haven't seen any extra failure on Windows with ES5.1, ES2015 and ES.Next version of test262.
> Tested on the latest master ( [7b00db4](https://github.com/jerryscript-project/jerryscript/commit/7b00db40799630709e6017ba9073815a4b6cb4eb) ) in cmd.exe not in PS.
> 
> (update: tested in PS too, got the same result)
> 
> There is only one es2015 test hang (infinite loop) in strict and non-strict mode, but it isn't related
> to Windows, but a spec change - language/statements/for-of/iterator-next-reference.js.

It's matched now

closed 2021-01-09 by lygstate