jerryscript-project/jerryscript #4387: very bigint add are failing

lygstate, 2020-12-29:
```
243593253485081309523243243243242342342344362643646254254354467873247493759274385743987528437958739817589345n+128371928731279387219372189738972189372198372198748932174836159763956392176593153213123123123123123123212312232323232323232323232323232323232323232323232331231231231231n
```

akosthekiss, 2020-12-29:
@lygstate If this is a bug report, please, use the bug report template and give all requested details (git revision, build platform, build command, test case, observed output, expected output, etc.)

In the current state of this issue, it is not clear what "failing" means. Filling in the template will help with that.

rerobika, 2020-12-29:
This issue seems invalid.

I've tested on the latest master (83f9c38) and there was no assertion nor incorrect result.

#### v8:
```js
128371928731279387219372189738972189372198372198748932174836403357209877257902676456366366365465465467574955878577486677700196479816991597618067219851670289971048820576n
```

#### jerry:
```js
128371928731279387219372189738972189372198372198748932174836403357209877257902676456366366365465465467574955878577486677700196479816991597618067219851670289971048820576
```

closed 2020-12-29 by rerobika

lygstate, 2020-12-29:
@rerobika  Which O you using?

lygstate, 2020-12-29:
> This issue seems invalid.
>
> I've tested on the latest master ([83f9c38](https://github.com/jerryscript-project/jerryscript/commit/83f9c38ed148a583ef34ea89047dd0e3bdecb37e)) and there was no assertion [...]

I guess you are not using the js shell to print the result?

rerobika, 2020-12-29:
I see now. This problem is independent from the OS you are using, it's simply the limitation of the REPL buffer size in the `main-unix.c`. 

My two suggestions.

1, Dump the test into a `.js` file and run it with the engine.
2, Modify the [JERRY_BUFFER_SIZE](https://github.com/jerryscript-project/jerryscript/blob/master/jerry-main/main-unix.c#L33) macro to enlarge the REPL buffer size according to your personal requirements.

akosthekiss, 2020-12-29:
@rerobika Should we treat this as a feature request? Should the REPL in jerry be able to work with arbitrarily-sized output?

lygstate, 2020-12-29:
both nodejs quickjs output have training `n`, we need make consistence with that.

akosthekiss, 2020-12-29:
Wait. We need to conform to ES specs. According to what I've read, the BigInt::toString abstract operation does not say anything about an n suffix. https://tc39.es/ecma262/#sec-numeric-types-bigint-tostring

That suffix is to denote BigInt literals in JS code.

As for Node.js, it behaves differently for `console.log(123n);` and `console.log(String(123n));`. The first snippet prints `123n`, while the second prints `123`. (If I get the docs of node.js right, `console.log` uses `util.format`, which in turn uses `util.inspect`, which "returns a string representation of object that is intended for debugging. The output of util.inspect may change at any time and should not be depended upon programmatically." https://nodejs.org/api/util.html#util_util_inspect_object_options) So, right now, I'm leaning towards "no n suffix -- JerryScript is right".