jerryscript-project/jerryscript #4406: Amalgamated JerryScript source release?

IngwiePhoenix, 2021-01-04:
Hello there!

I noticed the `ENABLE_ALL_IN_ONE` flags within `build.py` and wondered: Are those files intended to be released, and if, is there a stable way to obtain them as releases come out? I am working with a couple of amalgamated projects - sqlite, mongoose just to name a few - and I was wondering if there was such a release for JerryScript.

My intend is to create a [vlang](https://github.com/vlang/v) wrapper. In order to keep install/build sizes and times small, I was looking to find an amalgamated version that I could use in that wrapper.

Kind regards,
Ingwie

akosthekiss, 2021-01-04:
Hi @IngwiePhoenix , I think that it's not `ENABLE_ALL_IN_ONE` that you are after. (That cmake option is only useful during the build.) However, please, refer to https://jerryscript.net/configuration/#single-source-build-mode . That should help you more.

There is no "release" of these amalgamated files. However, you can grab a release (a tagged revision of the project) and run the command shown in the docs and get the files.

akosthekiss, 2021-01-11:
@IngwiePhoenix You might be interested to know that #4416 has landed.

IngwiePhoenix, 2021-01-11:
This is perfect! Thank you for the notification. Only just got done reinstalling a huge part of Windows (somehow winreg lost the visual studio data... so, had to go all the way through 14GB worth of reinstalls. woohoo...)

Will take a closer look at this. Again, thanks for the poke!

closed 2021-01-11 by IngwiePhoenix