jerryscript-project/jerryscript #4431: Build fails on s390x architecture

slyon, 2021-01-07:
###### JerryScript revision
The jerryscript git submodule was apparently upgraded from 
7cc9d65c095c6631bb547d863dfeadc470d8dfc6 (good)
to
0ef9b8e0273055fabe4a4665ca9bdee509ee4202 (bad)

###### Build platform
Ubuntu 21.04 (Linux 5.8.0-33-generic s390x)

###### Build steps
```sh
sbuild -d hirsute ./iotjs-1.0+715
```

###### Build log
```
[ 46%] Building C object jerry-core/CMakeFiles/jerry-core.dir/__/jerry-all-in.c.o
In file included from /usr/include/string.h:519,
                 from /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/jrt/jrt.h:20,
                 from /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/ecma/base/ecma-globals.h:20,
                 from /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/debugger/debugger.h:19,
                 from /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/api/jerry-debugger-transport.c:16,
                 from /<<BUILDDIR>>/iotjs-1.0+715/obj-s390x-linux-gnu/deps/jerry-host/jerry-all-in.c:1:
In function ‘memcpy’,
    inlined from ‘parser_stack_iterator_write’ at /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/parser/js/js-parser-mem.c:713:5,
    inlined from ‘parser_parse_if_statement_end’ at /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/parser/js/js-parser-statm.c:870:3,
    inlined from ‘parser_parse_statements’ at /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/parser/js/js-parser-statm.c:3273:15:
/usr/include/s390x-linux-gnu/bits/string_fortified.h:34:10: error: writing 16 bytes into a region of size 1 [-Werror=stringop-overflow=]
   34 |   return __builtin___memcpy_chk (__dest, __src, __len, __bos0 (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/jcontext/jcontext.h:32,
                 from /<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/api/jerry-debugger-transport.c:17,
                 from /<<BUILDDIR>>/iotjs-1.0+715/obj-s390x-linux-gnu/deps/jerry-host/jerry-all-in.c:1:
/<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/parser/js/js-parser-statm.c: In function ‘parser_parse_statements’:
/<<BUILDDIR>>/iotjs-1.0+715/deps/jerry/jerry-core/parser/js/js-parser-internal.h:317:11: note: at offset 0 to object ‘bytes’ with size 1 declared here
  317 |   uint8_t bytes[1];                           /**< memory bytes */
      |           ^~~~~
cc1: all warnings being treated as errors
make[6]: *** [jerry-core/CMakeFiles/jerry-core.dir/build.make:63: jerry-core/CMakeFiles/jerry-core.dir/__/jerry-all-in.c.o] Error 1
make[6]: Leaving directory '/<<BUILDDIR>>/iotjs-1.0+715/obj-s390x-linux-gnu/deps/jerry-host'
make[5]: *** [CMakeFiles/Makefile2:235: jerry-core/CMakeFiles/jerry-core.dir/all] Error 2
```

###### Test case
Give the JavaScript input that should be passed to the engine to trigger the
bug. Try and post a reduced test case that is minimally necessary to reproduce
the issue. As a rule of thumb, use Markdown's fenced code block syntax for the
test case. Attach the file (renamed to .txt) if the test case contains
'problematic' bytes that cannot be copied in the bug report directly.

###### Execution steps
Trying to build the deb package via sbuild on s390x architecture (works on other architectures).

###### Expected behavior
Jerryscript should build on  s390x as it does on other architectures.

akosthekiss, 2021-01-07:
@sylon Can you trigger the same build error without iotjs on latest jerryscript master? (The build command and the log show that iotjs is being built, not jerryscript alone.)

akosthekiss, 2021-01-07:
I've just found out that the reported revision is from July 2017! If the issue cannot be reproduced on latest master (or at least on a _much_ more recent revision), which I assume is the case, then this issue is invalid / can't fix / wont fix.

slyon, 2021-01-07:
Thanks for your quick response.
I tried to re-produce the problem by running `python3 tools/build.py` on `master` of jerryscript alone.
It works. Closing this issue.

slyon, 2021-02-03:
Actually, I had another look into this. And the Jerryscript revision in question is: 56e328be416c32d31988644a57e18aeb959e6be2

It builds successfully when I run `python3 tools/build.py` but it does fail with the above mentioned error when I try to build the iotjs package (with jerryscript dependency) via `debuild`. Debuild is using the CMake build system, not `tools/build.py`

I see there are some `TARGET_ARCH` definitions inside the build system. Is the s390x architecture supported at all, or should it be removed from Debian/Ubuntu? Does `tools/build.py` pass any special parameters compared to Debian's CMake build flags (https://sources.debian.org/src/iotjs/1.0+715-1/debian/rules/#L27)?

akosthekiss, 2021-02-04:
There shouldn't be any specific issue blocking the build for s390x. That being said, we have never built either JerryScript or IoT.js for that architecture. We simply don't have that kind of hardware.

As for the options passed to cmake by IoT.js's build.py, please, see here: https://github.com/jerryscript-project/iotjs/blob/master/tools/build.py#L318

Also, there is work going on to bump the JerryScript dependency in IoT.js to latest release (2.4.0) (https://github.com/jerryscript-project/iotjs/pull/1966). However, I'm not sure that it would solve this problem.

slyon, 2021-02-04:
Thanks for getting back to this!
I've now been able to reproduce the failure on 56e328b, using `python3 tools/build.py --debug --all-in-one ON`. It seems to be related to the `-D_FORTIFY_SOURCE=2` CPPFLAG on s390x. The default build optimizes for size `-Os` (CMake: MinSizeRel), which seems to disable the fortify/hardening, while Debian builds with `-O2`, triggering the error.

```
ubuntu@hh-jerryscript-bisect:~/iotjs/deps/jerry$ python3 tools/build.py --debug --all-in-one ON
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- CMAKE_BUILD_TYPE            Debug
-- CMAKE_C_COMPILER_ID         GNU
-- CMAKE_SYSTEM_NAME           Linux
-- CMAKE_SYSTEM_PROCESSOR      s390x
-- BUILD_SHARED_LIBS           OFF
-- ENABLE_LTO                  ON
-- ENABLE_STRIP                ON
-- JERRY_CMDLINE               ON
-- JERRY_CMDLINE_TEST          OFF
-- JERRY_CMDLINE_SNAPSHOT      OFF
-- JERRY_LIBFUZZER             OFF (FORCED BY COMPILER)
-- JERRY_PORT_DEFAULT          ON (FORCED BY CMDLINE OR LIBFUZZER OR TESTS)
-- JERRY_EXT                   ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_LIBM                  ON
-- UNITTESTS                   OFF
-- DOCTESTS                    OFF
-- ENABLE_ALL_IN_ONE              ON
-- JERRY_CPOINTER_32_BIT          OFF
-- JERRY_DEBUGGER                 OFF
-- JERRY_ERROR_MESSAGES           OFF
-- JERRY_EXTERNAL_CONTEXT         OFF
-- JERRY_PARSER                   ON
-- JERRY_LINE_INFO                OFF
-- JERRY_LOGGING                  OFF
-- JERRY_MEM_STATS                OFF
-- JERRY_MEM_GC_BEFORE_EACH_ALLOC OFF
-- JERRY_PARSER_DUMP_BYTE_CODE    OFF
-- JERRY_PROFILE                  es.next
-- JERRY_REGEXP_STRICT_MODE       OFF
-- JERRY_REGEXP_DUMP_BYTE_CODE    OFF
-- JERRY_SNAPSHOT_EXEC            OFF
-- JERRY_SNAPSHOT_SAVE            OFF
-- JERRY_SYSTEM_ALLOCATOR         OFF
-- JERRY_VALGRIND                 OFF
-- JERRY_VM_EXEC_STOP             OFF
-- JERRY_GLOBAL_HEAP_SIZE         (512)
-- JERRY_GC_LIMIT                 (0)
-- JERRY_STACK_LIMIT              (0)
-- JERRY_GC_MARK_LIMIT            (8)
-- FEATURE_INIT_FINI           OFF
-- Performing Test HAVE_TM_GMTOFF
-- Performing Test HAVE_TM_GMTOFF - Success
-- Looking for include file time.h
-- Looking for include file time.h - found
-- Looking for include file unistd.h
-- Looking for include file unistd.h - found
-- ENABLE_LINK_MAP             OFF
-- JERRY_TEST_STACK_MEASURE    OFF
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/iotjs/deps/jerry/build
Scanning dependencies of target jerry-libm
[  1%] Building C object jerry-libm/CMakeFiles/jerry-libm.dir/acosh.c.o
[  2%] Building C object jerry-libm/CMakeFiles/jerry-libm.dir/acos.c.o
[  4%] Building C object jerry-libm/CMakeFiles/jerry-libm.dir/asinh.c.o
[  5%] Building C object jerry-libm/CMakeFiles/jerry-libm.dir/asin.c.o
[  7%] Building C object jerry-libm/CMakeFiles/jerry-libm.dir/atan.c.o
[  8%] Building C object jerry-libm/CMakeFiles/jerry-libm.dir/atan2.c.o
[ 10%] Building C object jerry-libm/CMakeFiles/jerry-libm.dir/atanh.c.o
[... 67 lines cut ...]
[ 94%] Building C object jerry-main/CMakeFiles/jerry.dir/main-unix.c.o
[ 95%] Building C object jerry-main/CMakeFiles/jerry.dir/main-utils.c.o
[ 97%] Building C object jerry-main/CMakeFiles/jerry.dir/cli.c.o
[ 98%] Building C object jerry-main/CMakeFiles/jerry.dir/main-options.c.o
[100%] Linking C executable ../bin/jerry
In function ‘memcpy’,
    inlined from ‘parser_stack_push’ at ../jerry-core/./deps/jerry/jerry-core/parser/js/js-parser-mem.c:534:5,
    inlined from ‘scanner_push_class_declaration’ at ../jerry-core/./deps/jerry/jerry-core/parser/js/js-scanner-util.c:1478:3:
/usr/include/s390x-linux-gnu/bits/string_fortified.h:29:10: error: writing 8 bytes into a region of size 1 [-Werror=stringop-overflow=]
   29 |   return __builtin___memcpy_chk (__dest, __src, __len, __bos0 (__dest));
      |          ^
../jerry-core/./deps/jerry/jerry-core/parser/js/js-scanner-util.c: In function ‘scanner_push_class_declaration’:
../jerry-core/./deps/jerry/jerry-core/parser/js/js-parser-internal.h:317:11: note: at offset 0 to object ‘bytes’ with size 1 declared here
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/cc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
gmake[2]: *** [jerry-main/CMakeFiles/jerry.dir/build.make:153: bin/jerry] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:368: jerry-main/CMakeFiles/jerry.dir/all] Error 2
gmake: *** [Makefile:149: all] Error 2
==============================
Build failed with exit code: 2
==============================
```

slyon, 2021-02-04:
I've now worked around this problem by building the Debian/Ubuntu package using this option:
```
CMAKE_OPTIONS+=-DCMAKE_BUILD_TYPE=MinSizeRel
```

But it might still be good to look into this at some point, to fix it properly.

galpeter, 2021-02-04:
Is it possible to trigger this issue with the v2.4.0 release? ( https://github.com/jerryscript-project/iotjs/pull/1966 )

I've tried to recreate the build issue: I've created a debian chroot with s390x cross compiler (gcc 10.2.1), but was unable to trigger this. Also tried various combinations (like -D_FORTIFY_SOURCE=2, debug/release build, all-in-one, not all-in one, older/newer commit)

slyon, 2021-02-05:
I cannot seem to reproduce it on v2.4.0. Previously I could trigger it using `python3 tools/build.py --debug --all-in-one ON`. But the `--all-in-one` option is gone from v2.4.0, using `--amalgam ON` istead does not trigger the same problem (but another one, with `python != python3`):
```
ubuntu@hh-jerryscript-bisect:~/jerryscript$ python3 tools/build.py --debug --amalgam ON
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- CMAKE_BUILD_TYPE               Debug
-- CMAKE_C_COMPILER_ID            GNU
-- CMAKE_SYSTEM_NAME              Linux
-- CMAKE_SYSTEM_PROCESSOR         s390x
-- BUILD_SHARED_LIBS              OFF
-- ENABLE_AMALGAM                 ON
-- ENABLE_LTO                     ON
-- ENABLE_STRIP                   ON
-- ENABLE_COMPILE_COMMANDS        ON
-- JERRY_VERSION                  
-- JERRY_CMDLINE                  ON
-- JERRY_CMDLINE_TEST             OFF
-- JERRY_CMDLINE_SNAPSHOT         OFF
-- JERRY_LIBFUZZER                OFF (FORCED BY COMPILER)
-- JERRY_PORT_DEFAULT             ON (FORCED BY CMDLINE OR LIBFUZZER OR TESTS)
-- JERRY_EXT                      ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_MATH                     OFF
-- UNITTESTS                      OFF
-- DOCTESTS                       OFF
-- JERRY_CPOINTER_32_BIT          OFF
-- JERRY_DEBUGGER                 OFF
-- JERRY_ERROR_MESSAGES           OFF
-- JERRY_EXTERNAL_CONTEXT         OFF
-- JERRY_PARSER                   ON
-- JERRY_LINE_INFO                OFF
-- JERRY_LOGGING                  OFF
-- JERRY_MEM_STATS                OFF
-- JERRY_MEM_GC_BEFORE_EACH_ALLOC OFF
-- JERRY_PARSER_DUMP_BYTE_CODE    OFF
-- JERRY_PROFILE                  es.next
-- JERRY_REGEXP_STRICT_MODE       OFF
-- JERRY_REGEXP_DUMP_BYTE_CODE    OFF
-- JERRY_SNAPSHOT_EXEC            OFF
-- JERRY_SNAPSHOT_SAVE            OFF
-- JERRY_SYSTEM_ALLOCATOR         OFF
-- JERRY_VALGRIND                 OFF
-- JERRY_VM_EXEC_STOP             OFF
-- JERRY_GLOBAL_HEAP_SIZE         (512)
-- JERRY_GC_LIMIT                 (0)
-- JERRY_STACK_LIMIT              (0)
-- JERRY_GC_MARK_LIMIT            (8)
-- Looking for sin in m
-- Looking for sin in m - found
-- FEATURE_INIT_FINI              OFF
-- Performing Test HAVE_TM_GMTOFF
-- Performing Test HAVE_TM_GMTOFF - Success
-- Looking for include file time.h
-- Looking for include file time.h - found
-- Looking for include file unistd.h
-- Looking for include file unistd.h - found
-- ENABLE_LINK_MAP                OFF
-- JERRY_TEST_STACK_MEASURE       OFF
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/jerryscript/build
[  3%] Generating ../amalgam/jerryscript.c, ../amalgam/jerryscript.h
/bin/sh: 1: python: not found
gmake[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/build.make:415: amalgam/jerryscript.c] Error 127
gmake[1]: *** [CMakeFiles/Makefile2:206: jerry-core/CMakeFiles/jerry-core.dir/all] Error 2
gmake: *** [Makefile:149: all] Error 2
==============================
Build failed with exit code: 2
==============================
```

If I link python -> python3 the build passes.