jerryscript-project/jerryscript #4439: AddressSanitizer failed

owl337, 2021-01-11:
###### JerryScript revision
fdaacde6

###### Build platform
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g --strip=off \
--system-allocator=on --logging=on --linker-flag=-fuse-ld=gold \
--error-messages=on --profile=es2015-subset  
```

###### Test case
```
assert.sameValue('foo'.repeat(1031883772), '');

```

###### Output
```
==19606==ERROR: AddressSanitizer failed to allocate 0xb8841000 (-1199304704) bytes of LargeMmapAllocator (error code: 12)
==19606==Process memory map follows:
	0x1ffff000-0x24000000	
	0x24000000-0x28000000	
	0x28000000-0x40000000	
	0x56596000-0x567c6000	/root/jerryscript/build/bin/jerry
	0x567c6000-0x567c8000	/root/jerryscript/build/bin/jerry
	0x567c8000-0x567eb000	/root/jerryscript/build/bin/jerry
	0xf4e00000-0xf4f00000	
	0xf5000000-0xf5100000	
	0xf5200000-0xf5300000	
	0xf5400000-0xf5500000	
	0xf5600000-0xf5700000	
	0xf5800000-0xf5900000	
	0xf5a00000-0xf5b00000	
	0xf5c00000-0xf5d00000	
	0xf5e00000-0xf5f00000	
	0xf6000000-0xf6100000	
	0xf6200000-0xf6300000	
	0xf63dd000-0xf75f1000	
	0xf75f1000-0xf760d000	/usr/lib32/libgcc_s.so.1
	0xf760d000-0xf760e000	/usr/lib32/libgcc_s.so.1
	0xf760e000-0xf760f000	/usr/lib32/libgcc_s.so.1
	0xf760f000-0xf762a000	/lib32/libpthread-2.27.so
	0xf762a000-0xf762b000	/lib32/libpthread-2.27.so
	0xf762b000-0xf762c000	/lib32/libpthread-2.27.so
	0xf762c000-0xf762e000	
	0xf762e000-0xf7636000	/lib32/librt-2.27.so
	0xf7636000-0xf7637000	/lib32/librt-2.27.so
	0xf7637000-0xf7638000	/lib32/librt-2.27.so
	0xf7638000-0xf763b000	/lib32/libdl-2.27.so
	0xf763b000-0xf763c000	/lib32/libdl-2.27.so
	0xf763c000-0xf763d000	/lib32/libdl-2.27.so
	0xf763d000-0xf780f000	/lib32/libc-2.27.so
	0xf780f000-0xf7810000	/lib32/libc-2.27.so
	0xf7810000-0xf7812000	/lib32/libc-2.27.so
	0xf7812000-0xf7813000	/lib32/libc-2.27.so
	0xf7813000-0xf7816000	
	0xf7816000-0xf78df000	/lib32/libm-2.27.so
	0xf78df000-0xf78e0000	/lib32/libm-2.27.so
	0xf78e0000-0xf78e1000	/lib32/libm-2.27.so
	0xf78e1000-0xf7a4f000	/usr/lib32/libasan.so.4.0.0
	0xf7a4f000-0xf7a51000	/usr/lib32/libasan.so.4.0.0
	0xf7a51000-0xf7a54000	/usr/lib32/libasan.so.4.0.0
	0xf7a54000-0xf7eac000	
	0xf7eac000-0xf7ed8000	
	0xf7ed8000-0xf7edb000	[vvar]
	0xf7edb000-0xf7edd000	[vdso]
	0xf7edd000-0xf7f03000	/lib32/ld-2.27.so
	0xf7f03000-0xf7f04000	/lib32/ld-2.27.so
	0xf7f04000-0xf7f05000	/lib32/ld-2.27.so
	0xffcea000-0xffd0b000	[stack]
==19606==End of process memory map.
==19606==AddressSanitizer CHECK failed: ../../../../../src/libsanitizer/sanitizer_common/sanitizer_common.cc:118 "((0 && "unable to mmap")) != (0)" (0x0, 0x0)
    #0 0xf79d2eb1  (/usr/lib32/libasan.so.4+0xf1eb1)
    #1 0xf79f3f8b in __sanitizer::CheckFailed(char const*, int, char const*, unsigned long long, unsigned long long) (/usr/lib32/libasan.so.4+0x112f8b)
    #2 0xf79dc8da  (/usr/lib32/libasan.so.4+0xfb8da)
    #3 0xf79eac78  (/usr/lib32/libasan.so.4+0x109c78)
    #4 0xf79039e2  (/usr/lib32/libasan.so.4+0x229e2)
    #5 0xf78ffec6  (/usr/lib32/libasan.so.4+0x1eec6)
    #6 0xf79c6f1c in malloc (/usr/lib32/libasan.so.4+0xe5f1c)
    #7 0x5665efd4 in jmem_heap_alloc /root/jerryscript/jerry-core/jmem/jmem-heap.c:254
    #8 0x5665f0ce in jmem_heap_gc_and_alloc_block /root/jerryscript/jerry-core/jmem/jmem-heap.c:291
    #9 0x5665f161 in jmem_heap_alloc_block /root/jerryscript/jerry-core/jmem/jmem-heap.c:325
    #10 0x566f32f4 in ecma_builtin_string_prototype_object_repeat /root/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-string-prototype.c:1212
    #11 0x566f3efb in ecma_builtin_string_prototype_dispatch_routine /root/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-string-prototype.c:1499
    #12 0x56608710 in ecma_builtin_dispatch_routine /root/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1490
    #13 0x566089a6 in ecma_builtin_dispatch_call /root/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1522
    #14 0x5662ce87 in ecma_op_function_call_native /root/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1246
    #15 0x5662d810 in ecma_op_function_call /root/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1416
    #16 0x566a764b in opfunc_call /root/jerryscript/jerry-core/vm/vm.c:822
    #17 0x566c10af in vm_execute /root/jerryscript/jerry-core/vm/vm.c:4959
    #18 0x566c16aa in vm_run /root/jerryscript/jerry-core/vm/vm.c:5060
    #19 0x566a58bc in vm_run_global /root/jerryscript/jerry-core/vm/vm.c:350
    #20 0x565bc88a in jerry_run /root/jerryscript/jerry-core/api/jerry.c:608
    #21 0x565b5669 in main /root/jerryscript/jerry-main/main-unix.c:123
    #22 0xf7655f20 in __libc_start_main (/lib32/libc.so.6+0x18f20)
    #23 0x565b4ea0  (/root/jerryscript/build/bin/jerry+0x1eea0)

```

Credits: Found by chong from OWL337.

rerobika, 2021-01-11:
To decide whether it's a bug or not is quite hard task.

First of all, it happens only when the engine is compiled with ASAN . Our internal string limit is 2^32 - 1, so hypothetically we can represent that large string. In `String.prototype.repeat` we check whether the new string's size is in this range and try to allocate it. However until now on unsuccessful allocation the engine was terminated with `ERR: out of memory` so I've created #4452 to throw `RangeError` instead of terminating the engine (similar to V8). I think this is the most we can do. Of course we could start fine-tuning the max allocation limit, but I don't think it's worth it only for testing arbitrary size allocation with ASAN.

rerobika, 2021-01-18:
Summary: I'd rather say that's not the engines fault. In some sorf of stange situation where someone needs langer then 2GB and has enough memory the engine can handle it if not OOM abort is thrown.

closed 2021-01-18 by rerobika